KillerInk/FreeDcam #1: NGM Forward Art - missing dng, only bayer available

MasterPIC, 2017-11-03:
[log.txt](https://github.com/KillerInk/FreeDcam/files/1440675/log.txt)

Phone Model:
NGM Forward Art_MTK

FreeDcam Version:
Last release

Problem Description:
Only bayer (and jpeg) is available, dng is possible only by manually converting from bayer. 
On an older release the app offered also dng as output format by saving both jpeg and dng after taking a shot.

KillerInk, 2017-11-06:
I dont have a Device wich support Camera1 and Raw.
From your Log the DngProfile gets written. 
The Problem should be in the [PictureFormatHandler](https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/camera1/parameters/modes/PictureFormatHandler.java#L48-L72)

KillerInk, 2017-11-26:
alpha26 should work again

MasterPIC, 2017-11-26:
Yes, it works again.

Thanks
Armando

closed 2017-11-26 by MasterPIC