KillerInk/FreeDcam #2: Why isn't this app on the Play Store?

feinstein, 2017-11-17:
Well the title says it all...why can't I find this as an app on the Play store?

KillerInk, 2017-11-18:
There are various reasons:
- I'm to lazy to respond to Playstore Commit's and Emails.
- To many bugy/different Camera Api's. Some are impossible to add, without having that device.
- My bad english, or other's english is not existing and he/she write in their homelanguage.
- It's infact to powerfull for most People.
- Ignorant people
- And more stuff...

Coding is a hobby for me. But if it feels like work, the fun is gone.

feinstein, 2017-11-18:
So, can't you post on the Play Store and ignore all the dumb people out there? Hahaha

You can leave a message on the app's description saying "I don't give support to this App, don't expect me to reply to your emails"

KillerInk, 2017-11-26:
Its now published again.

closed 2017-11-26 by KillerInk

feinstein, 2017-11-26:
I can't find it, could you share the link?

KillerInk, 2017-11-27:
https://play.google.com/store/apps/details?id=troop.com.freedcam