KillerInk/FreeDcam #15: How you switch the dual camera of an Huawei P9 (mono/colour)?

beniroquai, 2018-01-08:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
Huawei P9

FreeDcam Version:
-----------------
latest

Problem Description:
--------------------
The app you're further developing is really really fantastic! I'm really impressed how much effort and time you put into it. 
I'm looking for a very basic way to get the monochromatic camera work with a very basic Camera API2 framework. I've seen this line: `HUAWEI_DUAL_SENSOR_MODE = getKeyType("com.huawei.capture.metadata.dualSensorMode", Byte.TYPE);` in the file https://github.com/KillerInk/FreeDcam/blob/59fdcd08b429ad6305d2c2982ef92d9c4a61059b/app/src/main/java/com/huawei/camera2ex/CaptureRequestEx.java

Is there any "easy" way to switch the camera channel? I don't find the line where you do that. I'm really appreciating your help! :) 
Finally I would like to simply build a super simple app which captures the BW video stream of the Huawei P9.  I need it for a project at school.

KillerInk, 2018-01-09:
class itself that set it, is more a wrapper:
https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/camera2/parameters/modes/DualCameraModeHuaweiApi2.java

The final parameter gets set due a CaptureRequest.
`CaptureRequest.set(HUAWEI_DUAL_SENSOR_MODE,(Byte) 0; `
0= primary color
1= secondary mono
2= dual

You can create the camera2 custom keys only with reflection because the ctor is not public.
But then it works like a normal camera2 key.

beniroquai, 2018-01-09:
> Antworten an: KillerInk/FreeDcam <reply@reply.github.com>

Wow! Thank you very much fort he fast answer! I had the feeling, that there is nothing too fancy about it – actually weird, that Huawei still didn’t make a API for this. 

If it is really that simple, I could basically take the RAW-camera from Googles github as a base and build an interface where I can address the line below, right? 

Or what exactly do you mean by “reflection”. I have to admit, that I’m not that familiar with android programming (not an expert at all..).

Are you aware of an easy interface of such a project? Your camera is awesome, but understanding it and bringing it to compile is a bit too challenging for me. Anyway our group is now using your APP form the playstore. Really great work. Awesome!!! Thanks!!

https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/camera2/parameters/modes/DualCameraModeHuaweiApi2.java

KillerInk, 2018-01-09:
Seh grad du hast ne deutsche email adresse.
Huawei benutzt ne versteckte api. Oder wie sie es nennen silent parameters.
Im prinzip brauchst du nur die 4 Dateien:
CaptureRequestEx
CaptureResultEx
CameracharacteristicsEx
RefelectionHelper

In freedcam selber dreht sich alles um den SettingsManager.
Der featuredetector füllt den settingsmanger mit den möglichen features\parametern. Der parameterhandler erstellt dann alles für die aktive api basierend auf den settings.

Mit reflection kann ma klassen instanzieren die privat sind oder nen privaten konstruktor haben oder nicht in deinem namespace zufinden sind.

Umzu verstehen wie camera2 funktioniert würd ich mir mal das camera2 sample von google anschaun.

Kurz zusammengefasst:
Camera öffnen,
Capturesession erstellen,
CaptureRequest erstellen,
Preview erstellen und starten.

beniroquai, 2018-01-09:
Wow! Klasse! Ich denke gleich mal drüber nach! Hast du ein bisschen Erfahrung mit Handykameras? Ich brauch das für ein Projekt hier im Institut. Vorprozessierungen wie die H264 komprimierung etc. werden wahrscheinlich auf dem Chip der Kamera direkt erledigt, oder? Welche Parameter hat man wohl noch in der Hand? Am besten wäre ein RAW-Monochromatic Video-stream. 

 

Woher kommst du, wenn ich fragen darf?

 

Viele Grüße! 

Datum: Dienstag, 9. Januar 2018 um 08:41

KillerInk, 2018-01-09:
Mir ist noch kein handy begegnet das yuv 444 unterstützt. In der regel kriegst yuv420 und wennst viel glück hast 422.
In der theory kannst mit camera2 30 raws in der sekunde aufnehmen wenns nen full device ist. Aber nich mitm p9, ist ein limited device durch die huawei api.

Zum recording selber.:
Im prinzip werden die preview frames zum mediarecorder weiter gereicht. Der übernimmt das konvertieren in h264.
Du kannst auch direkt mitm codec selber arbeiten aber das kann ziemlich handy specific werden. Da nicht jedes handy die selben codecs unterstützt. P9 unterstützt zb kein h265.
Dazu kommt noch ob der codec hardware unterstützung hat oder obs software seitig encoded werden muss. 

Ich selber bin aus münchen, nicht weit von der tum.

beniroquai, 2018-01-09:
Hey, vielen Dank schon einmal für die ganze Hilfe. Ich habe jetzt einmal angefangen das Camera2Video Beispiel zu modifizieren. Ich habe schon ein bisschen Erfahrung sammeln können, was das Cam2Api angeht. Ich bin allerdings gerade komplett überfragt wo ich dein Vorschläge einbauen kann. 

Ich denke irgendwo hier:
`/**
     * Start the camera preview.
     */
    private void startPreview() {
        if (null == mCameraDevice || !mTextureView.isAvailable() || null == mPreviewSize) {
            return;
        }
        try {
            closePreviewSession();
            SurfaceTexture texture = mTextureView.getSurfaceTexture();
            assert texture != null;
            texture.setDefaultBufferSize(mPreviewSize.getWidth(), mPreviewSize.getHeight());
            mPreviewBuilder = mCameraDevice.createCaptureRequest(CameraDevice.TEMPLATE_PREVIEW);
            mPreviewBuilder.set(HUAWEI_DUAL_SENSOR_MODE, (byte) 1);

            Surface previewSurface = new Surface(texture);
            mPreviewBuilder.addTarget(previewSurface);

            mCameraDevice.createCaptureSession(Collections.singletonList(previewSurface),
                    new CameraCaptureSession.StateCallback() {

                        @Override
                        public void onConfigured(@NonNull CameraCaptureSession session) {
                            mPreviewSession = session;
                            updatePreview();
                        }

                        @Override
                        public void onConfigureFailed(@NonNull CameraCaptureSession session) {
                            Activity activity = getActivity();
                            if (null != activity) {
                                Toast.makeText(activity, "Failed", Toast.LENGTH_SHORT).show();
                            }
                        }
                    }, mBackgroundHandler);
        } catch (CameraAccessException e) {
            e.printStackTrace();
        }
    }`

 { (zu finden [hier](https://github.com/googlesamples/android-Camera2Video/blob/master/Application/src/main/java/com/example/android/camera2video/Camera2VideoFragment.java#L489)

Welche Methode muss ich adressieren um den CaptureRequest von RGB auf BW umzustellen? 
Im Prinzip könnte ich einfach deine App benutzen, aber die hat einfach zu viele Parameter - was eigentlich super ist ;-) 
Können wir uns wohl auch per Mail austauschen? Meine müsstest du haben, oder? 

Ansonsten: beniroquai at hotmail.de

beniroquai, 2018-01-10:
Mal eine kurze Frage, können wir wohl mal telefonieren? Meine Mail Adresse wäre beniroquai at hotmail.de die Portierung mit deinen Tipps hat gut geklappt. Perfekt. Danke!! :)
Ich hab allerdings noch gefühlt 1000 fragen was die handykameras und die interne wirkweise angeht. 

Viele Grüße

KillerInk, 2018-01-13:
Hab dir ne email geschrieben, aber kam bisher nichts zurück.
Vom telefonieren bin ich kein großer fan. 
KillerInk@email.de falls die fragen noch bestehen^^

closed 2018-01-15 by KillerInk