KillerInk/FreeDcam #16: LG G2 does not have the option for DNG output

gojpdchx, 2018-01-22:
Phone Model: LG G2 (D802)
FreeDcam Version: 4.1 beta 5

**Steps to Reproduce**
1. Open FreeDcam app.
2. Click the "Picture Format Mode" button. (https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/res/drawable-hdpi/quck_set_format2.png)

**Actual Result**
Only `bayer` and `jpeg` are available.

**Expected Result**
In addition to `bayer` and `jpeg`, `dng` should also be available.
In version 3.3.147, [1] `dng` is available for this device.
I could not find any logs in the `DCIM/FreeDcam` folder, instead I found `dngprofiles.xml`.
`dngprofiles.xml`:
```
<devices>
<device name = "LG_G2">
<opcodes>
<camera id="0">
<opcode2></opcode2>
<opcode3>https://github.com/KillerInk/FreeDcam_opcodes/blob/master/LG_G2/opc3.bin?raw=true</opcode3>
</camera>
</opcodes>
<filesize size= "16224256">
<blacklvl>64</blacklvl>
<whitelvl>1023</whitelvl>
<width>4208</width>
<height>3082</height>
<rawtype>0</rawtype>
<colorpattern>bggr</colorpattern>
<rowsize>5264</rowsize>
<matrixset>Imx135</matrixset>
</filesize>
<filesize size= "16424960">
<blacklvl>64</blacklvl>
<whitelvl>1023</whitelvl>
<width>4212</width>
<height>3120</height>
<rawtype>0</rawtype>
<colorpattern>bggr</colorpattern>
<rowsize>5264</rowsize>
<matrixset>Imx135</matrixset>
</filesize>
</device>
</devices>
```

[1] Source code previously available at https://github.com/troop/FreeDcam.

KillerInk, 2018-01-23:
Hi cant reproduce it on my g3. it loads the config file and show dng as pictureformat.
pls check if a bayerformat is set. if not set it, stop freedcam, remove it from the tasks, restart it and hope that it shows it. 
a log get only written inside DCIM/FreeDcam when a log.txt is created.

greets

gojpdchx, 2018-01-24:
@KillerInk 
I reinstalled (updated) the newest version from Google Play to try again.
If I do the following, `dng` will not be available.
1. Go to Android Settings -> Apps -> FreeDcam and clear app data (thus making it a fresh installation).
2. Open FreeDcam and swipe to open the settings page.
3. The "default" setting of `Bayer Format`, i.e. `bayer-mipi-10bggr`, can be seen.

That "default setting" seems to be a placeholder because I have to click on "Bayer Format" and choose `bayer-mipi-10bggr`. My wild guess is that with FreeDcam at stock settings (e.g. fresh installation), there is no value assigned to "Bayer Format" (despite the fact that `bayer-mipi-10bggr` is shown). The user has to manually choose one to make `dng` available.

If you would allow me, I would like to ask some more questions in this issue instead of opening additional issues:
1. (At stock settings, have not changed any picture shooting mode or any other modes/parameters) DNG files produced have the suffix "_BURST0". How can I remove it? (I do not have this issue when using version 3.3.147.)
2. What are the differences between `bayer-mipi-10bggr`, `bayer-ideal-qcom-10bggr` and `bayer-qcom-10bggr`? (e.g. brightness and color)

KillerInk, 2018-01-24:
now the last part about different bayer formats.
[mipi(tight)](https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/pixfmt-srggb10p.html) and[ qcom(loose)](https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/pixfmt-srggb10dpcm8.html) are different packed.

the ideal should contain data with removed vignett and some other stuff. 
but on some devices that cause rainbow colors on the final image or just hang the camera. mipi is the format wich works on all devices.

gojpdchx, 2018-01-25:
Many thanks for the fix and explanation.

closed 2018-01-25 by gojpdchx