KillerInk/FreeDcam #17: FreeDcam comeback

letkan, 2018-02-01:
Phone Model: Xiaomi Mi 4c
---------------

FreeDcam Version: 4.1beta6

I'm really glad someone took over this project and for the first time almost all options work on my current phone and no crashes. I only tested camera1 API, will check camera2 later.

My only issue is with creating a slow motion video profile which I couldn't figure out. Also + and - and 30fps do nothing, I'm guessing it's work in progress.

KillerInk, 2018-02-02:
I dont have a mi4c so cant say much about. On my g3 highspeed recording works.
Maybe your device dont support (dont use video-hfr) it or its caused due LOS or something else is wrong or need to get triggered bevor recording. Some devices are forced to 30fps. For me its not a WIP.

letkan, 2018-02-04:
Video-hfr seems to be supported (see attachment), 120fps works in Snap camera and Miui camera. Actually Freedcam video recording is quite broken on this device, if you care to fix it and need something like logcat and stuff, I'll be glad to provide.
[camera-info3.txt](https://github.com/KillerInk/FreeDcam/files/1692624/camera-info3.txt)

closed 2018-03-12 by KillerInk