KillerInk/FreeDcam #19: How to add  z11 mini s to supported devices?

SpencerDexter, 2018-02-04:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model: NX549j
---------------

FreeDcam Version:4.1beta6
-----------------

Problem Description:How to add  z11 mini s to supported devices? How to set imx318 custom matrix?
--------------------

KillerInk, 2018-02-05:
to get dng support you have to create a device profile like that:

```
 <device name ="ZTE_Z5SMINI">
        <models>
            <item>NX403A</item>
        </models>
        <camera1>
            <nightmode>1</nightmode>
            <virtuallensfilter>true</virtuallensfilter>
        </camera1>
        <filesize size= "17522688"> <!-- filesize could be wrong -->
            <blacklvl>64</blacklvl>
            <whitelvl>1023</whitelvl>
            <width>4212</width>
            <height>3120</height>
            <rawtype>1</rawtype>
            <colorpattern>rggb</colorpattern>
            <rowsize>0</rowsize>
            <matrixset>G4</matrixset>
        </filesize>
    </device>
```
https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/res/raw/supported_devices.xml#L3646

or capture a bayer file and use (from the viewer) RawToDng to create the profile.

SpencerDexter, 2018-02-05:
Works well

closed 2018-02-05 by SpencerDexter