KillerInk/FreeDcam #20: Bracketing

SpencerDexter, 2018-02-04:
Phone Model:NX549J
---------------

FreeDcam Version:4.1beta6 
-----------------

Problem Description: Focus bracketing seems not  working. It makes many images at one focus distance and the last image at another. BTW exposure bracketing makes 3 images of the same exposure.
Stacking feature is great but almost useless without auto alignment.
--------------------

KillerInk, 2018-02-05:
then your device is a legacy device wich dont support manuals on camera2.
on last version i enabled camera2 too for such devices and it load api2 by default wich is wrong.
so af/ae bracket on cam2 dont work on legacy devices and get hidden on next update.

you can use the stacking feature only when you capture with a tripod. its good enough to merge a static scene^^ feel free to implement auto alignment^^

SpencerDexter, 2018-02-05:
camera2probe shows "Full" support of camera2 api

KillerInk, 2018-02-05:
then it should work. you set min max distances? on my p9 and nex5 its working.
image count can get set with the burst manual.
[image]

SpencerDexter, 2018-02-05:
I checked many times. If I set min10sm, max40sm it works like a charm (I used a ruler)
But if I set min10, max_infinity it makes 29 images of infinity and one of min10 (I used a bottle in front of the window and none of the 30 images got the bottle in focus)

KillerInk, 2018-02-05:
yes you are right. i added there a logic problem^^