KillerInk/FreeDcam #21: CameraParameters

DrRamm, 2018-02-04:
Phone Model:
Xiaomi Mi3W, 8.1 android
FreeDcam Version:
4.1 beta6
Problem Description:
Why FreeDcam doesnt read cameraparamets? Bcz i edited *.txt and there is no difference (new params are here https://gist.github.com/DrRamm/2d544e41c86b6fce011eea96c894e1d0)

KillerInk, 2018-02-04:
That file is only a dump from the cameraparameters. It read them direct from the camera when you run the featuredetector and fill the settings.

DrRamm, 2018-02-05:
ohhh how i can fix them? CameraWrapper should fix it

KillerInk, 2018-02-05:
Check if you are on camera1. Beta6 loads by default camera2 also for legacy devices. But thats fixed in source already.

Featuredetector fills the settingsmanager. Parameterhandler then create the supported modes based on the settingsmanager.
I dont see something new in your parameters. With switching to cam1 all problems should be gone.

closed 2018-02-17 by KillerInk