KillerInk/FreeDcam #23: Zero dng in camera1

SpencerDexter, 2018-02-05:
Phone Model: NX549J
FreeDcam Version: 4.1beta6

Problem Description: I entered own dngprofile that easily converts bayer. Shot some dng in camera1 fnd now it makes zero files instead of dng. What can be the reason?

--------------------

KillerInk, 2018-02-05:
dont know. maybe somethings wrong with the dngprofile

KillerInk, 2018-02-05:
a correct **dngprofiles.xml** look like that
```
<devices>
 <device name ="Huawei_HONOR5x">
<filesize size= "16424960">
            <blacklvl>64</blacklvl>
            <whitelvl>1023</whitelvl>
            <width>4208</width>
            <height>3120</height>
            <rawtype>0</rawtype>
            <colorpattern>rggb</colorpattern>
            <rowsize>5264</rowsize>
            <matrixset>IMX258_Qualcomm</matrixset>
        </filesize>
    </device>
</devices>
```

or capture a bayer and create it with the viewer and rawtodng, thats more fails save

SpencerDexter, 2018-02-05:
I created it with the viewer and rawdng. I even managed to make some dngs.

SpencerDexter, 2018-02-05:
It makes bayer of 1.64mb. I'm doing smth wrong.

SpencerDexter, 2018-02-05:
Camera1 makes bayer from 1 to 2 mb. Camera2 makes correct 43 mb bayer.

KillerInk, 2018-02-05:
1-2mb sounds like jpeg. try to rename fileending to .jpg 
maybe your device dont support rawoutput on camera1

SpencerDexter, 2018-02-05:
You are right.