KillerInk/FreeDcam #34: Xperias still have capped ISO

mirh, 2018-03-06:
I see you include the [right](https://github.com/LineageOS/android_device_sony_shinano-common/commit/0fad7808d04605b18caa089f2a718f0ac11e2de7) permission in your AndroidManifest.. 
But I still cannot select the "whole range" of settings from manual mode (XZ)

[Reportedly](https://forum.xda-developers.com/xperia-z5/themes-apps/app-ultimate-unlocked-camera-manual-t3486055/page10#post69474874), HAL may be disallowing some values when put in actual manual mode.

KillerInk, 2018-03-07:
I dont have such a device. Also its not clear what you understand as manual mode. Its open source, feel free to make your changes

KillerInk, 2018-03-07:
About what keys are you talking?
```
iso-speed-values
sony-iso-values
```

https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/featuredetector/Camera1FeatureDetectorTask.java#L1251-L1272

mirh, 2018-03-11:
I'm not sure how I'd be supposed to check that, 
Feature detection is just too quick to figure out. 
And *Save Camera Parameters* is not available in camera2 api, and doesn't report it for camera1. 

I was talking about the manual settings slider [from the UI](https://raw.githubusercontent.com/KillerInk/FreeDcam/master/playstoreimages/cameraui.png). 
Oddly, when I'm not forcing anything, it can show something like ISO 5200 - though final photo still shows to be capped at 3200 in exif.

KillerInk, 2018-03-12:
I dont think thats a problem from freedcam. its more the problem that the hal lock the iso to the analog range. 
Camera parameters are no more needed on camera2.
There the keys, if there are custom, need to get extracted from the app itself.
Also camera2 dont have two iso values that can get set.
When ae is in auto mode it decide wich iso get used.

That behavior is only on camera1 for ae
stock android
```
iso-values=auto,ISO_HJR,ISO100,ISO200,ISO400,ISO800,ISO1600,ISO3200
iso=auto
```

sony extension
```
sony-iso-values=auto,auto,100,200,400,800,1600,3200,auto
sony-iso=auto
```

the permission is needed that freedcam can load that lib
`System.loadLibrary("cameraextensionjni");`
https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/com/sonyericsson/cameraextension/CameraExtension.java
https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/com/sonyericsson/cameracommon/device/CameraExtensionValues.java

as said, i dont have such device. without concret explaination what todo its useless to continue.

mirh, 2018-03-12:
[Mhh, I see the clusterf*](https://forum.xda-developers.com/crossdevice-dev/sony/dev-porting-z3-camera-to-previous-t2897316). 
Will report back when I'll be able to get better logs.

mirh, 2018-04-21:
I have now a XZ2 compact
And I can totally force ISO 12800 from stock camera

FreeDcam instead just stops to 3190

KillerInk, 2018-04-22:
freedcam use that values returned from the camera. on third party apps you will get that as max iso.
maybe they hardcoded it in the stock app. to look deeper into it i need the stock app **deodexed**

mirh, 2018-04-22:
I.. Didn't seem to notice odex files for the relevant apks, so I just [attached](https://github.com/KillerInk/FreeDcam/files/1935278/CameraApks.zip) them plainly

Interesting permissions (for starters) seem: `com.sonymobile.cameracommon.permission` and `com.sonymobile.permission.SOMC_CAMERA`

KillerInk, 2018-04-22:
its harcoded

com.sonyericsson.android.camera.configuration.parameters.Iso

```
ISO_AUTO(-1, R.string.cam_strings_image_quality_control_shutter_speed_auto_txt, "auto", -1),
    ISO_50(-1, R.string.cam_strings_iso_50_txt, CameraParameters.AE_MODE_ISO_PRIO, 50),
    ISO_100(-1, R.string.cam_strings_iso_100_txt, CameraParameters.AE_MODE_ISO_PRIO, 100),
    ISO_200(-1, R.string.cam_strings_iso_200_txt, CameraParameters.AE_MODE_ISO_PRIO, 200),
    ISO_400(-1, R.string.cam_strings_iso_400_txt, CameraParameters.AE_MODE_ISO_PRIO, 400),
    ISO_800(-1, R.string.cam_strings_iso_800_txt, CameraParameters.AE_MODE_ISO_PRIO, MediaRecorder.MEDIA_RECORDER_INFO_MAX_DURATION_REACHED),
    ISO_1600(-1, R.string.cam_strings_iso_1600_txt, CameraParameters.AE_MODE_ISO_PRIO, Constants.WAITING_TIME_STOP_REC),
    ISO_3200(-1, R.string.cam_strings_iso_3200_txt, CameraParameters.AE_MODE_ISO_PRIO, SizeConstants.WIDTH_5_8MP_WIDE),
    ISO_6400(-1, R.string.cam_strings_iso_6400_txt, CameraParameters.AE_MODE_ISO_PRIO, 6400),
    ISO_12800(-1, R.string.cam_strings_iso_12800_txt, CameraParameters.AE_MODE_ISO_PRIO, 12800),
    ISO_25600(-1, R.string.cam_strings_iso_25600_txt, CameraParameters.AE_MODE_ISO_PRIO, 25600),
    ISO_51200(-1, R.string.cam_strings_iso_51200_txt, CameraParameters.AE_MODE_ISO_PRIO, 51200);
```

KillerInk, 2018-04-22:
can you pls attach the cameraparameters from freedcam

mirh, 2018-04-22:
[Here you are](https://github.com/KillerInk/FreeDcam/files/1935841/H8314_CameraParameters.txt) (taken of course after selecting camera1 api)

mirh, 2018-12-26:
Ehrm, putting aside we are back to square one then..
Why does the ISO gauge (on auto) say 5120 on photo shooting, but then the actual image/exif has 2884?

KillerInk, 2018-12-27:
thats caused due the different needs for the preview and image capture.
in preview its equal if its noise due high iso aslong it can focus fast. on a image capture ae may decided thats enough light is availible to lower iso and use a slower shutterspeed to reduce noise.

mirh, 2018-12-27:
Mhh, would make sense, if maybe counterintuitive 

But really, not even in pitch dark it seems to raise above 2884

mirh, 2020-11-27:
And now (4.3.17) any kind of manual iso makes for a completely black picture and preview.

KillerInk, 2020-11-27:
Check your shutter speed if its not set to 1/100000

mirh, 2020-11-30:
You were right (why does it fix itself when you touch ISO?). 
All the previous problems are still there though.. 
ISO gauge is quite off the charts. 
Auto mode maxes out at ISO 2884. 
And even in manual mode ISO can't go beyond 3200.

KillerInk, 2020-12-01:
i dont know. looks like sony limited it.

closed 2020-12-03 by KillerInk

KillerInk, 2026-04-18:
5years later... Love it 😄