KillerInk/FreeDcam #35: Green color issue when recording 4k

oluja1973, 2018-03-07:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
Xiaomi Redmi 5 Plus

FreeDcam Version:
-----------------
4.1beta9

Problem Description:
--------------------
Hello,

first thing: This ist the best camera app in the appstore for devices with camera 1 api for video recording.
There are 2 bif reason for this.
1. You canreduce or increase brightnes and/or saturation for the video. This gives the video a flat profile which is great to color grade.
2. the image stabilisation works phenomenal. No shaky videos anymore. And this for each smartphone without eis or ois. Great work.

Now my problem. When activating 4k profile and enabling image stabilisation during video recording, i get green lines. See screenshot. Any idea what can this bee? If you could fixing this, this would be great. So i can record movies in 4k with the great features of freedcam.

Thx for your help and go on with this awesome piese of software.

Regards, Mario!

[image]

KillerInk, 2018-03-08:
Hi.
I don't think its fixable. Its only thought to get used with **Image** capture.
It works on some devices with  >= 1080. 
That happens due the Mediarecorder. it configure on recording start the preview format. Imagestabilisation does not support the input that get in and produce this output.

greets

oluja1973, 2018-03-08:
Hello,

ok thanks for your feedback and the explanation. Go on with this great video recorder.
Thx for this piece of software.

Regards

KillerInk, 2018-03-11:
from google docs:
https://developer.android.com/reference/android/hardware/camera2/CaptureRequest.html#CONTROL_VIDEO_STABILIZATION_MODE

> In addition, not all recording sizes or frame rates may be supported for stabilization by a device that reports stabilization support. It is guaranteed that an output targeting a MediaRecorder or MediaCodec will be stabilized if the recording resolution is less than or equal to 1920 x 1080 (width less than or equal to 1920, height less than or equal to 1080), and the recording frame rate is less than or equal to 30fps.

closed 2018-03-12 by KillerInk