KillerInk/FreeDcam #43: HUAWEI_AVAILABLE_DUAL_PRIMARY parameters 0, 2, 3 instead of 0,1,2

mydri, 2018-04-16:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
Huawei Honor 9 Lite

FreeDcam Version:
-----------------
4.1beta9

Problem Description:
--------------------
The Huawei Honor 9 Lite  is equip with 13mp + 2mp camera. No mono camera.

Usually with mono camera, the parameters for HUAWEI_AVAILABLE_DUAL_PRIMARY are
0= primary color 
1= secondary mono
2= dual

Since Honor 9 Lite does not have mono camera, The parameters are 0, 2, 3. Maybe 
3 = secondary color (2mp)

Could you please add that parameter in the menu selection of the software.

Thank you

KillerInk, 2018-04-16:
hi.
About wich api are you talking? 1 or 2? If about cam1 pls attach the cameraparameters
Im not sure if the honor9L realy use that parameter. From what i know that gets only used for devices with mono cam support.
In your case the second(2mp) camera should be, sofar, always active to get the focusmap.

mydri, 2018-04-16:
Hi,

Thank you for your quick reply. Sorry I did not mention earlier. I am referring to API2.
I am currently trying to write a software that can capture and save individual photos from main camera (13mp) and secondary camera(2mp) separately.  

I have written the codes to get the characteristics of that Honor 9 lite. Here is part of the code:

`        for (int i=0; i< numOfCam; i++) {

            txtAllcam.append(" \n camera " + i + " LENS_FACING: " + characteristicsCam[i].get(CameraCharacteristics.LENS_FACING));

            if (Build.MANUFACTURER.equals("HUAWEI")) {
                txtAllcam.append(" \n\nHUAWEI Specific only *** START *** \n");
                final Byte b = characteristicsCam[i].get(CameraCharacteristicsEx.HUAWEI_MONO_MODE_SUPPORTED);
                if (b != null) {
                    txtAllcam.append(" \n camera " + i + " Mono Mode Supported: " + " TRUE " + " \n");
                } else if (b == null) {
                    txtAllcam.append(" \n camera " + i + " Mono Mode Supported: " + " FALSE " + " \n");
                }

                final byte[] array = characteristicsCam[i].get(CameraCharacteristicsEx.HUAWEI_AVAILABLE_DUAL_PRIMARY);
                if (array != null) {
                    int availmodes =  characteristicsCam[i].get(CameraCharacteristicsEx.HUAWEI_AVAILABLE_DUAL_PRIMARY).length;
                    txtAllcam.append(" \n camera " + i + " Dual Primary Available: " + " TRUE.   Length: " + availmodes + " \n");
                    for (int k=0; k< availmodes ; k++)
                        txtAllcam.append("\n value: " + characteristicsCam[i].get(CameraCharacteristicsEx.HUAWEI_AVAILABLE_DUAL_PRIMARY)[k] + " ");

                } else if (array == null) {
                    txtAllcam.append(" \n camera " + i + " Dual Primary Available: " + " FALSE " + " \n");
                }
          
                txtAllcam.append(" \n\n HUAWEI Specific only  *** END *** \n \n");
            }
        }
`
Here is the sceenshot of the charateristics:

[image]

It shows that the parameters I got are 0, 2, 3.
I assume that parameter 3 for acessing the secondary camera (2mp). Am I wrong?

Thanks.

KillerInk, 2018-04-16:
freedcam ignores on feature detect the availible modes but use the correct values.
so in your case:

primary color = 0;
secondary mono = 2;
dual = 3;
you should be able to switch it.

mydri, 2018-04-17:
> freedcam ignores on feature detect the availible modes but use the correct values.
so in your case:

>primary color = 0;
secondary mono = 2;
dual = 3;
you should be able to switch it.

Ok. thanks for the explanation.  
Only "primary color" works perfectly.
If I choose "secondary mono", I do not get anything from the camera. only black screen.
If I choose "dual", I get display on the screen, but I cannot capture it (pressing the shutter button).

> the output from 2mp could be:
https://developer.android.com/reference/android/graphics/ImageFormat.html#DEPTH16
or
https://developer.android.com/reference/android/graphics/ImageFormat.html#DEPTH_POINT_CLOUD
or also someother format specifyed by huawei.

Thanks for the info. I will investigate futher about this.

KillerInk, 2018-04-19:
> If I choose "secondary mono", I do not get anything from the camera. only black screen.

i think thats caused due the 2mp cam, it dont support yuv output.

> If I choose "dual", I get display on the screen, but I cannot capture it (pressing the shutter button).

think its the same reason. for me that sound like the 2mp cam only creates the focusmap.
that combination may work only with jpeg.

closed 2018-08-30 by KillerInk