KillerInk/FreeDcam #44: Mono mode doesn't work in full resolution on Huawei Mate 9.

opensourcemattress, 2018-04-25:
Phone Model: Huawei Mate 9 
---------------

FreeDcam Version: 4.1.beta9 (from git)
-----------------

Problem Description:
--------------------

I can capture mono on Mate 9 but there is no full resolution for mono in settings. Mate 9 has different sensors with different resolution (20Mp mono and 12Mp color)  but app settings always shows color resolutions. 
I tried to set full resolution in code in dumb way (just added line  ```valueToSet = "5120x3840";``` in ```SetValue``` of PictureSizeModeApi2) but after that there is only black screen and nothing happens.

Please, help me to solve this issue.

opensourcemattress, 2018-04-26:
Thanks for answer!

I tried to change resolution here https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/featuredetector/Camera2FeatureDetectorTask.java#L693 . Not sure how to check current sensor so I added all available resolutions from mono sensor without check, just to see  whether it will work. After that I was able to select needed resolution in settings and I got 20Mp jpeg, so partially it works.  What doesn't work is dng/bayer. It saves corrupted 12Mp dng (all black). I missed something or it's because some huawei restrictoins?

I also found interesting strings in `/vendor/` directory (simply made `grep -a camera * -R`). There are  `camera.sdk.open` and `android.hw.dualCameraVerificationMode` . I suppose that they are android properties which can be set with setprop. May be it will be useful.

KillerInk, 2018-04-26:
you can dump camera with adb
`adb shell dumpsys media.camera > cam.txt`

android.hw.dualCameraVerificationMode
gets used to check if one of the cameras is overlayed by something
it can get check when a captureresult returns, with HUAWEI_DUAL_SENSOR_ABORMAL.

>  What doesn't work is dng/bayer. It saves corrupted 12Mp dng (all black).

i think it because RAW_SENSOR size returned ist that for the color cam.
https://github.com/KillerInk/FreeDcam/commit/5fa50e01f992886a391bc81618712825e431d37d#diff-acd365dcb233ff3d3c019376c3ff611fR275
not sure if it work. but maybe hardcoding the size will work

opensourcemattress, 2018-04-30:
Thanks for commit and answer!

I attach dumped camera info: [cam.txt](https://github.com/KillerInk/FreeDcam/files/1961979/cam.txt)

In my mate 9 I get sizes in ascending order (so I need to take last size for largest size, not first), I not sure about other huawei phones. May be it's better to group result into pairs and then sort them?

I tried to take a picture and got error somewhere in native part of DNG saver. I got same error when I tried to take picture by myself in simple prototype. 

BUT when I switched to custom DNG saver I took my first mono .dng from Mate 9!

So, It works with custom DNG saver. I can't say that I have good understanding of whole app but I suppose that it's better to add sizes detection somewhere in https://github.com/KillerInk/FreeDcam/blob/master/app/src/main/java/freed/cam/apis/featuredetector/Camera2FeatureDetectorTask.java , so user can select largest available size for monochrome in jpeg scenario. But I still not sure how to check current setting (mono or color) in this file. 
I tried to check current sensor as in your commit but I got empty string in `SettingsManager.get(SettingKeys.dualPrimaryCameraMode).get()`, so I can't compare it to anything.

opensourcemattress, 2018-04-30:
And what about DUAL mode?
Can I take picture from both sensors simultaneously?

KillerInk, 2018-05-01:
> I attach dumped camera info: cam.txt

thx

> In my mate 9 I get sizes in ascending order (so I need to take last size for largest size, not first), I not sure about other huawei phones. May be it's better to group result into pairs and then sort them?

for sure it would be the better approach. i just added it that way to check if it works.
better would be, in case of raw, to take hwSubActiveArraySize.
the settingsmanager stores the capbilites detected. depending on that(if supported) the parameters gets loaded in ParameterHandlerApi2. 
raw itself use the full sensor size, but for you color cam there is binning support.

> I tried to take a picture and got error somewhere in native part of DNG saver. I got same error when I tried to take picture by myself in simple prototype.

that normaly happens when something is wrong with the returned metadata.
The stock dngconverter is less fail save then the freedcam converter.

> And what about DUAL mode?
> Can I take picture from both sensors simultaneously?

for jpeg it should work. for raw i cant confirm. 
Huawei worked around camera2 that it works more like camera1. they added a top interface to the lower api2 interface. Normaly that mono cam should get returned as a own (like front) camera to open. But thats not the case.
that merging stuff happens on native side. Also you cant check if truly both cameras trigger on a dual capture. on p9 it happens only on a dark scene and the focus must close to infinity.

KillerInk, 2018-05-02:
46ad557f1505b615f103ad055349be9afeb28b08 should handel now the secondary sizes and raw too.

opensourcemattress, 2018-05-02:
I compiled it, started app and tried to set up it and take a shot. Now it crashes before launching:

```
05-02 18:29:26.640 7432-7644/troop.com.freedcam E/AndroidRuntime: FATAL EXCEPTION: CameraFragmentManager
    Process: troop.com.freedcam, PID: 7432
    java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equals(java.lang.Object)' on a null object reference
        at freed.cam.apis.camera2.modules.PictureModuleApi2.setHuaweiOutput(PictureModuleApi2.java:314)
        at freed.cam.apis.camera2.modules.PictureModuleApi2.setOutputSizes(PictureModuleApi2.java:244)
        at freed.cam.apis.camera2.modules.PictureModuleApi2.startPreview(PictureModuleApi2.java:158)
        at freed.cam.apis.camera2.modules.PictureModuleApi2.InitModule(PictureModuleApi2.java:119)
        at freed.cam.apis.basecamera.modules.ModuleHandlerAbstract.setModule(ModuleHandlerAbstract.java:125)
        at freed.cam.apis.camera2.Camera2Fragment.initCamera(Camera2Fragment.java:288)
        at freed.cam.apis.basecamera.MainToCameraHandler.handleMessage(MainToCameraHandler.java:90)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:156)
        at android.os.HandlerThread.run(HandlerThread.java:61)
```

opensourcemattress, 2018-05-02:
Seems like I just forgot to set 'Secondary size' and there is no default value.

KillerInk, 2018-05-02:
try latest commit it should work now

KillerInk, 2018-05-03:
yes i change the subarray to sencondaryjpeg sizes to have something to test.

```
android.scaler.availableJpegSizes (d0003): int32[24]
        [320 240 640 480 ]
        [1280 720 1920 1080 ]
        [1536 1536 3264 1840 ]
        [3264 2448 3840 2160 ]
        [3968 2976 2048 1536 ]
        [2976 2976 3968 2240 ]
      com.huawei.device.capabilities.secondAvailableJpegSizes (80000082): int32[24]
        [320 240 640 480 ]
        [1280 720 1920 1080 ]
        [1536 1536 3264 1840 ]
        [3264 2448 3840 2160 ]
        [3968 2976 2048 1536 ]
        [2976 2976 3968 2240 ]
      com.huawei.device.capabilities.hwSubSensorJpegSize (80000085): int32[26]
        [320 240 640 480 ]
        [1280 720 1920 1080 ]
        [3264 1840 3264 2448 ]
        [3840 2160 3968 2976 ]
        [2048 1536 2976 2976 ]
        [3968 2240 5120 2880 ]
        [5120 3840 ]
```
when changes are made to the feature detector you have to rerun it that the settings get updated

licaon-kter, 2018-11-06:
@KillerInk So these are integrated and I should see a "Secondary size" option and be able to select a 20Mp resolution on Mate 9? Because I don't have this option on 4.1beta9.

Running OmniROM now, but using the Huawei camera I can capture 20Mp mono.

licaon-kter, 2018-11-06:
Could this be reopened?

Tested HEAD too, Mate 9 & OmniROM 8.1.

Clean app, I only select "Mono" and the resolution, and I get:
* 320..3264 -> correct size
* 3840 -> image frozen
* 3968, 2048, 2976, 3968 -> correct size
* 5120 -> image size 1920x1080

KillerInk, 2018-11-07:
sure, i did not get a response back so i closed it after some months.
```
 com.huawei.device.capabilities.hwSubSensorJpegSize (80000085): int32[26]
        [320 240 640 480 ] = 320x240 , 640x480
        [1280 720 1920 1080 ] = 1280x720, 1920x1080
        [3264 1840 3264 2448 ] = .....
        [3840 2160 3968 2976 ] = .....
        [2048 1536 2976 2976 ] = .....
        [3968 2240 5120 2880 ] = .....
        [5120 3840 ]                   = 5210x3840
```

just to make sure how the array have to get read

my self only have a  p9, and both back cameras have same size.

there the secondaray sensor size gets read.
https://github.com/KillerInk/FreeDcam/blob/7214b3f9fe0ee5e77c5a43dfb083ec7166fdd0d6/app/src/main/java/freed/cam/apis/featuredetector/Camera2FeatureDetectorTask.java#L498

and here the preview output get choosen
https://github.com/KillerInk/FreeDcam/blob/7214b3f9fe0ee5e77c5a43dfb083ec7166fdd0d6/app/src/main/java/freed/cam/apis/camera2/modules/helper/FindOutputHelper.java#L26

KillerInk, 2018-11-07:
In theory the code should work. I have no idea why it freez on some resolutions.
The preview size gets choosen depending on screensize and aspectratio from image or movie. I dont know whats wrong.

licaon-kter, 2019-01-10:
Any thoughts?

KillerInk, 2019-01-11:
not realy. would need that device to find out whats causing the freez.

Julientaming, 2020-10-01:
> Thanks for commit and answer!
>
> I attach dumped camera info: [cam.txt](https://github.com/KillerInk/FreeDcam/files/1961979/cam.txt)
>
> In my mate 9 I get sizes in ascending order (so I need to take [...]

Sorry how did you do that with customdng ??

KillerInk, 2020-10-02:
bevor you can work with custom dng you need to get bayer capture working. its the binary raw dump from sensor.
aslong it freez on capture or produce 0 byte bayers this wont help you much

Julientaming, 2020-10-03:
Can I see a sample of raw with the mate 9 please ?

closed 2020-12-03 by KillerInk