KillerInk/FreeDcam #45: Issues found on Huawei P9 with FreeDcam

konyzulphrea, 2018-04-26:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
Huawei P9 (EVA-L29)

FreeDcam Version:
4.1 Beta9
-----------------

Problem Description:
--------------------
There is no white Balance function
It seems there is no suitable color matrix so doing little more job in mono dng
There is an issue when shooting in 60fps h264

KillerInk, 2018-04-26:
> There is no white Balance function. 

There is but its not working with images only with video capture.
you have to change the tonemap to CONTRAST_CURVE. then a new button appears in the manual holder.

> It seems there is no suitable color matrix so doing little more job in mono dng

Whats wrong with the stock one? In case you use freedcam converter and have a custom profile active make sure the colormatrix is set to off. that way the stock matrix get used.

> There is an issue when shooting in 60fps h264

What is the issue? up to 1920x1080 60fps works for me with AE active. if you deactivate AE frames drop to 30fps. but thats the normal behavior. only ae can break the 60 fps wall.

konyzulphrea, 2018-04-26:
Thank you for replying some of my questions. I think It's time to try the
contrast setting to look further. My colormatrix is set off.

closed 2018-08-30 by KillerInk