KillerInk/FreeDcam #46: Capture from both sensors simultaneously (Huawei)

opensourcemattress, 2018-05-02:
Are there way to capture from 2 sensors simultaneously?

I tried to add second CaptureRequestHandler without preview but I got error "Session has been closed; further changes are illegal". I created separate session for mono and seems like when I use it default session is closes.

KillerInk, 2018-05-03:
thats caused by how huawei implemented their api. you can open each camera only into one capturesession.
what would work is to use front and back camera at same time.
for us the mono and color cam have the same id and can only get used in the same capturesession.
so no.

KillerInk, 2018-05-03:
there is a small problem for your device, due the fact that the mono and color sensor dont have the same size.
dualPrimaryCameraMode is now forced to recreate the jpeg/rawImageReader when you switch between them(that happen due the stopPreviewAsync/startPreviewAsync -> wich trigger the picturemodule to get recreated)

opensourcemattress, 2018-05-03:
It's fine even if it will take some time. But how can I guarantee that all settings were applied to `CaptureRequestBuilder` and camera is ready with new settings?

KillerInk, 2018-05-04:
that is able to capture both images but reset focus position
just add it to the ModuleHandlerApi2

```
 else if (SettingsManager.getInstance().getFrameWork() == Frameworks.HuaweiCamera2Ex)
        {
            AeBracketApi2 aeBracketApi2 = new HuaweiAeBracketApi2(cameraUiWrapper,mBackgroundHandler,mainHandler);
            moduleList.put(aeBracketApi2.ModuleName(),aeBracketApi2);
            HuaweiColorMonoModuleApi2 colorMonoModuleApi2 = new HuaweiColorMonoModuleApi2(cameraUiWrapper,mBackgroundHandler,mainHandler);
            moduleList.put(colorMonoModuleApi2.ModuleName(), colorMonoModuleApi2);
        }
```

```
package freed.cam.apis.camera2.modules;

import android.os.Build;
import android.os.Handler;
import android.support.annotation.RequiresApi;
import android.widget.TextView;

import com.huawei.camera2ex.CaptureRequestEx;
import com.troop.freedcam.R;

import freed.cam.apis.basecamera.CameraWrapperInterface;
import freed.settings.SettingKeys;
import freed.settings.SettingsManager;

@RequiresApi(api = Build.VERSION_CODES.LOLLIPOP)
public class HuaweiColorMonoModuleApi2 extends PictureModuleApi2 {

    private CamColorModes camColorMode = CamColorModes.color;

    private enum CamColorModes
    {
        color,
        mono
    }

    public HuaweiColorMonoModuleApi2(CameraWrapperInterface cameraUiWrapper, Handler mBackgroundHandler, Handler mainHandler) {
        super(cameraUiWrapper, mBackgroundHandler, mainHandler);
        name = "Dual";
    }

    @Override
    public String ShortName() {
        return "Dual";
    }

    @Override
    public String LongName() {
        return "Dual";
    }

    @Override
    protected void finishCapture() {
        super.finishCapture();
        if (camColorMode == CamColorModes.color)
        {
            SettingsManager.get(SettingKeys.dualPrimaryCameraMode).set(SettingsManager.getInstance().getResString(R.string.hw_dualcamera_Secondary));
            camColorMode = CamColorModes.mono;
            DestroyModule();
            InitModule();
            cameraUiWrapper.captureSessionHandler.SetParameterRepeating(CaptureRequestEx.HUAWEI_DUAL_SENSOR_MODE, Byte.valueOf((byte) 2), true);
            mBackgroundHandler.postDelayed(()-> DoWork(), 500);
            
        }
        else {
            SettingsManager.get(SettingKeys.dualPrimaryCameraMode).set(SettingsManager.getInstance().getResString(R.string.hw_dualcamera_Primary));
            camColorMode = CamColorModes.color;
            DestroyModule();
            InitModule();
            cameraUiWrapper.captureSessionHandler.SetParameterRepeating(CaptureRequestEx.HUAWEI_DUAL_SENSOR_MODE, Byte.valueOf((byte) 0), true);
        }

    }
}

```

KillerInk, 2018-05-04:
a different approach would be to register to the CamerValuesChangedCallback onFirstFrame Event.
And wait till it gets called and then do the capture. But the focus problem is the same

closed 2018-08-30 by KillerInk