KillerInk/FreeDcam #49: White/Black Frame

Smallchopper48, 2018-06-19:
Phone Model:
Xiaomi Redmi 3s Miui 9.5

FreeDcam Version:
4.1beta9

Problem Description:
In dng mode, there will be random white/black frame if you take a photo, for example if you take 3 photos, one of them will be only white/black frame, how to fix this? to overcome this i always took 2 photos, but sometimes that 2 photos will be appear as white frame

Smallchopper48, 2018-06-20:
im using camera1, and the manual dont work on my device so i always use auto, please help:(

Smallchopper48, 2018-06-20:
and one more thing, when i select dng, and set auto focus to for example continous autofocus, when i take photo it will reset to auto which is very inconvenient

Smallchopper48, 2018-06-20:
hey, good news is i just found the culprit, the thing is i always set the iso to 100 even in dng mode(my phone doesnt support manual on dng) and that aparently it cause the white/black frame, so thank u

and i tried the rawdumpinterface thingy because u said it will enable my manual control on dng, but that just make the app freeze when take a picture, so how exactly i can enable my manual setting on dng? because it's always on auto;/

KillerInk, 2018-06-21:
i dont think you will get it working with miui. for me it looks like that the blobs have problems.
i know there are redmi3s users but they use aosp based roms. think they use blobs from another device.
that also cause your non manual with dng problem

closed 2018-08-30 by KillerInk