KillerInk/FreeDcam #51: DNG Profile for Xiaomi Redmi Note 5A

LosTigeros, 2018-07-28:
**Phone Model:**
Xiaomi Redmi Note 5A

**FreeDcam Version:**
4.1_beta9

**Problem Description:**
There is no problem. Just giving you dng profile for this phone:
```
<devices>
  <device name="Redmi Note 5A">
    <opcodes>
    </opcodes>
    <filesize size="16510976">
      <blacklvl>64</blacklvl>
      <whitelvl>1023</whitelvl>
      <width>4208</width>
      <height>3120</height>
      <rawtype>3</rawtype>
      <colorpattern>grbg</colorpattern>
      <rowsize>0</rowsize>
      <matrixset>S5K3L8_Qualcomm</matrixset>
    </filesize>
  </device>
</devices>
```

You can close it when you read it ;)

KillerInk, 2018-07-30:
hi i need also the device model. else its not possible for freedcam to identfy the device

LosTigeros, 2018-07-30:
First two lines from camera parameters file:
```
Redmi Note 5A
3.18.31-perf-g33ac87a
```

commit d1949ca7 (2018-07-30): add Xiaomi Redmi Note 5A close #51
closed 2018-07-30 by KillerInk