KillerInk/FreeDcam #53: Infinity IIS - can't choose focus

LosTigeros, 2018-08-26:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
MyPhone Infinity IIS

FreeDcam Version:
4.1beta9

Problem Description:
I can't choose focus in app now.
[image]

Camera Parameters:
[INFINITY_II_S_CameraParameters.txt](https://github.com/KillerInk/FreeDcam/files/2321375/INFINITY_II_S_CameraParameters.txt)

Error in log:
```
D:10.45.30:AbstractParameterHandler:FocusManualMTK
java.lang.NumberFormatException: Invalid int: "auto"
	at java.lang.Integer.invalidInt(Integer.java:138)
	at java.lang.Integer.parse(Integer.java:410)
	at java.lang.Integer.parseInt(Integer.java:367)
	at java.lang.Integer.parseInt(Integer.java:334)
	at freed.cam.apis.basecamera.parameters.AbstractParameterHandler.setManualMode(AbstractParameterHandler.java:233)
	at freed.cam.apis.basecamera.parameters.AbstractParameterHandler.setManualSettingsToParameters(AbstractParameterHandler.java:168)
	at freed.cam.apis.camera1.Camera1Fragment.onPreviewOpen(Camera1Fragment.java:179)
	at freed.cam.apis.basecamera.CameraFragmentAbstract.handelMainMessage(CameraFragmentAbstract.java:268)
	at freed.cam.apis.camera1.Camera1Fragment.handelMainMessage(Camera1Fragment.java:547)
	at freed.cam.apis.basecamera.CameraToMainHandler.handleMessage(CameraToMainHandler.java:42)
	at android.os.Handler.dispatchMessage(Handler.java:111)
	at android.os.Looper.loop(Looper.java:194)
	at android.app.ActivityThread.main(ActivityThread.java:5637)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:959)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:754)
	at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:107)
```

KillerInk, 2018-08-26:
Try to rerun the featuredetector. Mtk stuff is hardcoded for mf and that can happen when you switch between different versions

LosTigeros, 2018-08-26:
Still same thing.

KillerInk, 2018-08-26:
from the pic it seems that the values are emtpy. due that you cant change it^^
the crash itself is legit, because of it get not applied to the camera if its set to auto.
i know its lazy coding but it works :D

LosTigeros, 2018-08-26:
Could you send me the latest build? I can't build it myself now because I'm on vacations :/

KillerInk, 2018-08-27:
sure 
https://github.com/KillerInk/FreeDcam/releases/tag/4.1final