KillerInk/FreeDcam #54: Crash while playing with ISO

mirh, 2018-08-28:
Phone Model:
XZ2 Compact

FreeDcam Version:
4.1final

Problem Description:
Title. As for the logs, I think you should see them from the google play console or something.

KillerInk, 2018-08-29:
```
java.lang.IllegalStateException: 
  at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError (CameraDeviceImpl.java:2182)
  at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest (CameraDeviceImpl.java:927)
  at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest (CameraDeviceImpl.java:974)
  at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest (CameraCaptureSessionImpl.java:238)
  at freed.cam.apis.camera2.CaptureSessionHandler.StartRepeatingCaptureSession (CaptureSessionHandler.java:392)
  at freed.cam.apis.camera2.CaptureSessionHandler.SetParameterRepeating (CaptureSessionHandler.java:519)
  at freed.cam.apis.camera2.parameters.ae.AeManagerCamera2.setIso (AeManagerCamera2.java:71)
  at freed.cam.apis.basecamera.parameters.ae.AeManager$ManualIso.setValue (AeManager.java:133)
  at freed.cam.apis.basecamera.parameters.AbstractParameter$BackHandler.handleMessage (AbstractParameter.java:113)
  at android.os.Handler.dispatchMessage (Handler.java:105)
  at android.os.Looper.loop (Looper.java:251)
  at android.os.HandlerThread.run (HandlerThread.java:65)
```

seems to happen only on sony device 🗡

KillerInk, 2018-08-29:
how often does it happen? happen it randomly or can you specific say "at this point its crashing"?

mirh, 2018-08-29:
Seems assured at least as soon as I fumble with the ISO roller

mirh, 2018-08-30:
Opencamera doesn't seem to have problems. 
Besides, I just noticed (silly me) that crash only happens when you exceed the ~3200 threshold. 

[dump.txt](https://github.com/KillerInk/FreeDcam/files/2336097/dump.txt)

KillerInk, 2018-08-30:
hm i see the problem, iso get overwritten due the xml. 
dont ask me why, but its set to max iso 51200

KillerInk, 2018-08-30:
i added it with #34

mirh, 2018-08-30:
Uh, yeah. Why was it? I don't remember that in the log (and as I said, both stock camera and [marketed spec](https://www.sonymobile.com/global-en/products/phones/xperia-xz2-compact/specifications/) is 12800)
Now I just need a new binary /s

mirh, 2018-12-26:
Fixed in 4.2

closed 2018-12-26 by mirh