KillerInk/FreeDcam #55: Honor 6A () hangs whilst saving raw

falket, 2018-08-30:
Phone Model:
---------------
Honor 6A

Camera Model Name               : DLI-L22
Software                        : DLI-L22-user 7.0 HONORDLI-L22 C432B127 release-keys

FreeDcam Version:
-----------------

FreeDCam_v4.1final

Problem Description:
--------------------

I have tried to take pictures enabling raw (dng 10 bit, dng 16 bit, bayer, raw+jpeg) but most of the time when i press the shutter white circle the screen update hangs and i don't get the file.

I've been anyway able to get some files, you can find them on wetransfer here: 

I don't have a "InternalSD/DCIM/FreeDcam/log.txt" file.

Suggestions welcome. Saving dng works with opencamera but the interface of FreeDcam seems very interesting to me ;) .

P.s. meanwhile i've played with various setting, but i haven't found a way to reset all of them to defaults.

Thank you in advance and keep up the great work!

KillerInk, 2018-08-31:
Hi, i dont think that the rawcapture itself freez. 
Its caused due the ae af precapture. 
To confirm it pls set focus to manual and same for iso and shutter speed.

About the settings. I personal dont have a need to reset settings to a default. Thats because the default would  be the values it gets from the featuredetector. And that are most time not optimal. But to reset it you can rerun the featuredetector.

About log. When you create in that path a log.txt it gets filled while freedcam run. But it can get big if you forget to delete it.

Greets

falket, 2018-08-31:
Thank you.

Unfortunatly.... i have disable everything and it worked. Then i re-enabled everything and it still works... I really don't understand why. Really disappointed, sorry. Maybe due to phone shutdown and restart meanwhile, who knows... but i did this it also before posting the request.

Btw, is there a support forum for this app? I would like to better understand some features.

Thank you again and keep up the great work.

KillerInk, 2018-09-01:
Would be interessting what caused it. I have an idea, pls reset freedcam to a clean install.
you can do it on your phone in settings/apps/freedcam clear cache & data. 
Now you should be at the state where the raw capture again freez. (if it now also works i have no idea what the problem was.)
open the settings in freedcam, and on the right panel scroll to bottom and set the dual camera. now the freez should be gone.

sorry there is no support forum due my laziness. There was one, but i shut i down because it got not used.
But you can ask your questions here.
the readme contains a small [Q&A](https://github.com/KillerInk/FreeDcam#qa)
And there is a bad filled [wiki](https://github.com/KillerInk/FreeDcam/wiki)

greets

falket, 2018-09-02:
I have tried but my phone doesn't have a dual camera, nor i see that option. And... it still works!

I think the problem was something about the destination directory. If i choose external and a folder, it creates a DCIM and a FreeDcam directory, maybe the first time i've choosed a parent folder with bad permissions. Would also be cool to have a subfolder for each day, imho (mtp transfer else is incredibly slow with a folder full of images).

Already took a look at Q&A and the wiki, i'll post here if i have questions. 

Regarding the forum, many foss (RawTherapee, Darktable, Gimp...) have a section at pixls.us, maybe you can find it interesting ( https://discuss.pixls.us/t/welcome-to-pixls-us-discussion/8 ).

Thank you again.

closed 2018-11-15 by KillerInk