KillerInk/FreeDcam #56: Select different audio sources

poutrosgit, 2018-09-03:
Currently there is no way to select an audio source and so we are stuck with the default one. However, the default one is too close to my camera and whenever the camera is on, it produces a noise that is very audible in the video. This can be fixed by changing the audio source so I think the feature is useful.

KillerInk, 2018-09-03:
the mediarecorder is very limited with its audio source input.
i had it set to camcorder some time ago and it crashed on some devices, due that freedcam is now using the default wich works on all devices.

https://developer.android.com/reference/android/media/MediaRecorder.AudioSource

i have first to check if its working as expected bevor i add it

KillerInk, 2018-11-15:
https://github.com/KillerInk/FreeDcam/pull/69

closed 2018-11-15 by KillerInk