KillerInk/FreeDcam #57: Camera resets when selecting an option for NighOverlay or Horizontal Line while Preview PostProcessing is enabled

poutrosgit, 2018-09-03:
The camera runs the feature detector and resets all values when Preview PostProcessing is enabled and any option is selected in NightOverlay or Horizontal Line.

KillerInk, 2018-09-03:
Hi,
i think you are using the legacy mode button, not the preview postprocessing one.
for legacy mode its needed to rerun the featuredetector, because, if support, the parameters from the camera change.
But you are right with the nightoverlay. it triggers the featuredetector. 
horizontal line seems total broken.
thx, i will look into it

poutrosgit, 2018-09-03:
NightOverlay and Horizontal Line only seem to trigger feature detection and reset when PostProcessing is enabled. Without postprocessing enabled, they don't trigger it. I don't know if they actually work because I haven't tested it. It gets triggered when I was selecting off in those features but only with postprocessing enabled.

closed 2020-12-03 by KillerInk