KillerInk/FreeDcam #59: List of bugs on LG V20

enomat, 2018-09-09:
Phone Model:
---------------
LG V20 H990DS
Android 7.0

FreeDcam Version:
-----------------
4.1final

Problem Description:
--------------------
After a fresh install:

1. Can't take photos if I choose manual ISO setting. I press the shutter button, but nothing happens. (Sometimes after I pressed the shutter button with manual ISO setting, then I switch back to auto ISO, it will take the photo at the moment when I switch back to auto ISO. It is like it recognizes that the shutter key was pressed only when I switch back to auto ISO.) The FreeDcam 4.1_beta9 did not have this bug.
2. The camera view is small (regression from the previous, 4.1_beta9 version)
3. The exposure compensation is set to -2.0 by default.
4. The highest ISO option available in manual settings is only 800. The LG V20's stock camera app has ISO settings up to 3200, so the phone supports it.
5. The highest exposure time available in manual settings is only 19 seconds. The LG V20 supports 30 seconds.
6. If I choose "Off" in the "Optical Image Stabilisation" setting the FreeDcam app will not focus anymore (neither the manual and auto focus works). I have to run FeatureDetector or clear app data to reset settings to get it working again, because the "On" option to revert this setting is missing.
7. There is no log file in "InternalSD/DCIM/FreeDcam/log.txt".

[image]

[image]

KillerInk, 2018-09-11:
Hi sorry for late respons. 

> Can't take photos if I choose manual ISO setting. I press the shutter button, but nothing happens. (Sometimes after I pressed the shutter button with manual ISO setting, then I switch back to auto [...]

i think more this problem is related to the ae af trigger. pls try to set everything to manual like focus, iso and exposure time and it should take pics. that is already fixed in source.

> The camera view is small (regression from the previous, 4.1_beta9 version)

yes that happen on some devices. but should be fixed too.

> The exposure compensation is set to -2.0 by default.

yes but only on first clean start, once you changed it, it should stick,

> The highest ISO option available in manual settings is only 800. The LG V20's stock camera app has ISO settings up to 3200, so the phone supports it.
> The highest exposure time available in manual settings is only 19 seconds. The LG V20 supports 30 seconds.

Well its lg. They dont allow 3rd partys to use higher iso and lower shutterspeeds. That problem is persists since they added camera2 to the g4.

> If I choose "Off" in the "Optical Image Stabilisation" setting the FreeDcam app will not focus anymore (neither the manual and auto focus works). I have to run FeatureDetector or clear app data to reset settings to get it working again, because the "On" option to revert this setting is missing.

its also lgs fault i think. normaly ois should return 0/1 off/on. but its possible to work around i think.

when you have time you can try to dump the camera with adb. it may return some hidden keys

```
adb shell
dumpsys media.camera > sdcard/cam.txt
```

greets

enomat, 2018-10-06:
Unfortunately I don't have adb and the phone drivers installed on my PC so I can't dump the camera. But I've found this page which has the dump files for V20 and other LG phones like G6 and V30.

https://forum.xda-developers.com/lg-v30/themes/cstark27-google-camera-mod-wide-angle-t3747263/page140

The specific comment on that page for V20:

https://forum.xda-developers.com/showpost.php?s=98801af9dfa34b817aae93ccfc5abed1&p=77539571&postcount=1399

I hope this is what you are looking for. I hope this will help you.

KillerInk, 2018-12-03:
sorry that there nothing happen. but i have no idea what todo. 
there are no extra keys added. i must have such a device to fix it.

enomat, 2018-12-03:
> i think more this problem is related to the ae af trigger. pls try to set everything to manual like focus, iso and exposure time and it should take pics. that is already fixed in source.

Unfortunately it is still not fixed. If I set ISO to 200, and the exposure time to 4, then it will take photos. But If I set the exposure time to 5 or higher, the green counter on the shutter key will never stop and it will never take photo. If I touch anything the app will crash. The FreeDcam 4.1_beta9 did not have this bug.

[image]

closed 2020-12-03 by KillerInk