KillerInk/FreeDcam #60: DNG resolution is only 4016x3008

milancheee, 2018-10-09:
**Phone Model:** Xiaomi Redmi Note 5 AI (codename : whyred)
---------------

**FreeDcam Version:** 4.1final
-----------------

**Problem Description:** 
After some tinkering with dngprofile I finally got FreeDcam working on Redmi Note 5, except one thing, resolution is 4016x3008, and it is supposed to be 4032x3024.

In Cameraparameters there is this line:

raw-size=4032x3024

Also Gcam with camera2 API enabled outputs this resolution. Also S5K2L7 datasheet lists this resolution: https://www.samsung.com/semiconductor/image-sensor/mobile-image-sensor/S5K2L7/

Is there anyway to get full resolution on Freedcam using camera1 API?

Also I'm not getting .jpg file when I shoot dng file, I used to get both dng+jpg when I was using FreeDcam on my Redmi Note 2.
--------------------

KillerInk, 2018-10-10:
hi
the size is ok. the orginal data have 4032x3024, but the dng doc says a cropping area should get defined because the raw data can have pixels for black calibration or simliar stuff.
so the stored rawdata have 4032x3024 but gets croppend in post processing to 4016x3008.

that your note2 can capture both is caused by mtk. its use a different approach then on snap socs.
by default mtk takes a jpeg, but you can set a path where the raw data gets dumped to the filesytem.
From there it gets read back in memory and processed to a dng

on snap soc its different. there the raw and jpeg gets returned in the same picturecallback.
you would have to take two pictures to get a jpeg and raw.
jpeg and raw at same time is only possible on camera2.

milancheee, 2018-10-10:
Alright, thank you for your answer and thank you for developing FreeDcam!

closed 2018-10-10 by milancheee