KillerInk/FreeDcam #61: Video recording do not starts after new update 4.1final

aunqur, 2018-10-17:
Phone Model: Mi A1
---------------

FreeDcam Version: 4.1final
-----------------

Problem Description: 
Every time i start camera and try to record video it shows either of following two errors. 
1. Start Recording failed Attempt to read from field 'long freed.utils.VideoMediaaProfile.MaxRecordingTime' on a null object reference
2. Start recording failed null.

Very occasionally it start recording though.

Recording was ok on previous version 4.1beta9
--------------------

aunqur, 2018-10-25:
I've traced error. sometimes recorder.currentVideoProfile is null and somtimes giving error 
java.lang.IllegalStateException. you can see it by setting debug at freed/cam/apis/camera1/modules/AbstractVideoModule.java:153 and see the Exception

KillerInk, 2018-10-26:
im not realy sure why it happen. i will look whats going on the next days

aunqur, 2018-10-28:
Thanks, video recording is now working fine. :)

closed 2018-10-29 by KillerInk