KillerInk/FreeDcam #62: DNG file is not opening in Google Photos, SnapSeed or Photoshop Express

aunqur, 2018-10-17:
Phone Model: Mi A1
---------------

FreeDcam Version: 4.1beta9
-----------------

Problem Description:
--------------------
I can view the dng file from FreedCam Viewer but not opening in other apps. Google Photos failed to show and do not sync with drive. SnapSeed and Adobe Photoshop Express doesnt open the dng file. Photoshop Express gives error "The original file appears to be unsupported or damaged" 

But I can open this https://www.kenrockwell.com/trips/2009-10/images/L1004220.DNG dng file in SnapSeed and Photoshop Express.

KillerInk, 2018-10-18:
Hi
wich api are you using? cam1/2?
It would be better to upload the not working dng file 📦 

But i think there is something wrong with the dng profile.
https://github.com/KillerInk/FreeDcam/blob/cc18996386116f8ea16586e97081c59e6bc752e1/app/src/main/res/raw/supported_devices.xml#L2396-L2400

aunqur, 2018-10-30:
Will it help if i set the white level from raw to dng tool in Viewer.  

But Latest version 4.1final from play store  is not capturing bayer also in my latest local build. It just shows timer in shutter button and freazes. 

I am using camera1 api.

KillerInk, 2018-10-30:
only reason why it could fail is normaly caused by ZSL (zeroshutterlag), it must get set to off

KillerInk, 2018-10-30:
in your internalStorage/DCIM/FreeDcam/config/dngprofiles.xml are the active profiles stored.
with fixing up the whitelvl 1023 the file can get open in every programm.
also 64 is the correct blacklvl.

aunqur, 2018-10-30:
Sweet. Works like a charm. 
bayer files are capturing after reset the app maybe was related to ZSL.

I've configured the dngprofiles.xml with values you have provided, Now dng captures and opening. 

any further suggestion on fine tuning the dngprofiles.xml. what is rawtype? it is 0 for mine. also what about rowsize?

aunqur, 2018-10-30:
Following settings will cause error while dng capture
1. Geo-Tag Location: on ( crashes )
2. RDI: enabled ( captures dng but freezes preview very often )
3. ZSL: on ( freezes on capture. nothing captures and timer run continues )

KillerInk, 2018-10-30:
> what is rawtype? it is 0 for mine

it tells the dng writer how to handel the inputdata, like on 10bit mipi the 5th byte contains bits for the other 4byte

https://github.com/KillerInk/FreeDcam/blob/2321e8b6eafd6671d502c0a01e4e1df8815ec047/app/src/main/java/freed/dng/DngProfile.java#L30-L41

https://github.com/KillerInk/FreeDcam/blob/2321e8b6eafd6671d502c0a01e4e1df8815ec047/app/src/main/jni/freedcam/DngWriter.cpp#L657-L684

--------------------------------------
> also what about rowsize?

with row size the dng can get trimmed, some dngs contains extra data and the true size is greater as the calculated. with that it can get overwriten. but its nomore needed due the crop tag

>     Geo-Tag Location: on ( crashes ) 

have to check, do not realy use it

>     RDI: enabled ( captures dng but freezes preview very often )

raw dump interface is needed on some 2016/2017 devices to get manuals working with rawcapture
if you have working manuals just keep it off

> 
    ZSL: on ( freezes on capture. nothing captures and timer run continues )

yes thats due the nature of zero shutter lag. i dont support a raw stream

closed 2018-10-30 by aunqur