KillerInk/FreeDcam #63: NDK error

Rudloff, 2018-10-24:
Hello,

I'm getting this error when trying to build 4.1final with `gradle assemble`:
```
Execution failed for task ':app:generateJsonModelDebug'.
> Build command failed.
  Error while executing process /usr/lib/android-ndk/ndk-build with arguments {NDK_PROJECT_PATH=null APP_BUILD_SCRIPT=/home/pierre/www/FreeDcam/app/src/main/jni/Android.mk NDK_APPLICATION_MK=/home/pierre/www/FreeDcam/app/src/main/jni/Application.mk APP_ABI=armeabi-v7a NDK_ALL_ABIS=armeabi-v7a NDK_DEBUG=1 APP_PLATFORM=android-14 NDK_OUT=/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj NDK_LIBS_OUT=/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib -j4 APP_SHORT_COMMANDS=false LOCAL_SHORT_COMMANDS=false -B -n}
  /usr/lib/android-ndk/build/core/build-binary.mk:472: warning: overriding recipe for target '/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/jpeg/jidctint.o'
  /usr/lib/android-ndk/build/core/build-binary.mk:472: warning: ignoring old recipe for target '/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/jpeg/jidctint.o'
  /usr/lib/android-ndk/build/core/build-binary.mk:472: warning: overriding recipe for target '/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/jpeg/jidctfst.o'
  /usr/lib/android-ndk/build/core/build-binary.mk:472: warning: ignoring old recipe for target '/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/jpeg/jidctfst.o'
  Android NDK: WARNING:/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/Android.mk:raw: LOCAL_LDLIBS is always ignored for static libraries    
  Android NDK: WARNING:/home/pierre/www/FreeDcam/app/src/main/jni/tiff/Android.mk:tiff: LOCAL_LDLIBS is always ignored for static libraries    
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/lib*.so
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  mkdir -p /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a
  echo [armeabi-v7a] "Gdbserver      ": "[arm-linux-androideabi-4.9] /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver"
  install -p /usr/lib/android-ndk/prebuilt/android-arm/gdbserver/gdbserver /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver
  echo [armeabi-v7a] "Gdbsetup       ": "/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup"
  echo "set solib-search-path /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a" > /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  echo "directory /usr/lib/android-ndk/platforms/android-14/arch-arm/usr/include null/tiff/libtiff/ ../../libjpeg/jpeg-9b/ /home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw /usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include /usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/libs/armeabi-v7a/include /usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include/backward /home/pierre/www/FreeDcam/app/src/main/jni/freedcam /home/pierre/www/FreeDcam/app/src/main/jni/libjpeg/jpeg-9c/include /home/pierre/www/FreeDcam/app/src/main/jni/tiff/libtiff /home/pierre/www/FreeDcam/app/src/main/jni/tiff/../libjpeg/jpeg-9b/ /home/pierre/www/FreeDcam/app/src/main/jni/tiff/../jpeg/" >> /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  mkdir -p /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debumga/kobej: *** No ru/lle to make targocet '/haome/pil/errea/www/FrmreeeDcam/apabip/sr-c/mavin/j7a/objs-debug/freedcam
  nie/frecedcam/Opchode.cpop',  [armeabi-v7a] "Compineeded by l'/ehom+e/pi+erre/ww tw/FreeDhcaum/mapbp/build/inte"rmediat:es/nd k"Build/dferbuge/obej/locald/arcmeabi-v7a/objs-adembu g/fr<ee=dc RawaTom/DOpncogde..ocp'.  Stop.
  p"
  /usr/lib/android-ndk/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/arm-linux-androideabi-g++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/RawToDng.o.d -fpic -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes  -g -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp  -fno-exceptions -fno-rtti -mthumb -O0 -UNDEBUG  -Inull/tiff/libtiff/ -I../../libjpeg/jpeg-9b/ -I/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/libs/armeabi-v7a/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include/backward -I/home/pierre/www/FreeDcam/app/src/main/jni/freedcam -DANDROID -fopenmp -Wa,--noexecstack -Wformat -Werror=format-security -fopenmp -Wno-c++11-narrowing -std=c++11 -fpic -frtti -fexceptions     -isystem /usr/lib/android-ndk/platforms/android-14/arch-arm/usr/include -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/RawToDng.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/RawToDng.o 
  echo [armeabi-v7a] "Compile++ thumb": "freedcam <= librawutils.cpp"
  /usr/lib/android-ndk/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/arm-linux-androideabi-g++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/librawutils.o.d -fpic -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes  -g -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp  -fno-exceptions -fno-rtti -mthumb -O0 -UNDEBUG  -Inull/tiff/libtiff/ -I../../libjpeg/jpeg-9b/ -I/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/libs/armeabi-v7a/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include/backward -I/home/pierre/www/FreeDcam/app/src/main/jni/freedcam -DANDROID -fopenmp -Wa,--noexecstack -Wformat -Werror=format-security -fopenmp -Wno-c++11-narrowing -std=c++11 -fpic -frtti -fexceptions     -isystem /usr/lib/android-ndk/platforms/android-14/arch-arm/usr/include -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/librawutils.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/librawutils.o 
  echo [armeabi-v7a] "Compile++ thumb": "freedcam <= DngWriter.cpp"
  /usr/lib/android-ndk/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/arm-linux-androideabi-g++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/DngWriter.o.d -fpic -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes  -g -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp  -fno-exceptions -fno-rtti -mthumb -O0 -UNDEBUG  -Inull/tiff/libtiff/ -I../../libjpeg/jpeg-9b/ -I/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/libs/armeabi-v7a/include -I/usr/lib/android-ndk/sources/cxx-stl/gnu-libstdc++/4.9/include/backward -I/home/pierre/www/FreeDcam/app/src/main/jni/freedcam -DANDROID -fopenmp -Wa,--noexecstack -Wformat -Werror=format-security -fopenmp -Wno-c++11-narrowing -std=c++11 -fpic -frtti -fexceptions     -isystem /usr/lib/android-ndk/platforms/android-14/arch-arm/usr/include -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/DngWriter.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/DngWriter.o 
  echo [armeabi-v7a] "Compile++ thumb": "freedcam <= DngStacker.cpp"
  /usr/lib/android-ndk/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/arm-linux-androideabi-g++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/DngStacker.o.d -fpic -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes  -g -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp  -fno-exceptions -fno-rtti -mthumb -O0 -UNDEBUG  -Inull/tiff/libtiff/ -I../../libjpeg/jpeg-9b/ -I/home/ [... cut ...] backward -I/home/pierre/www/FreeDcam/app/src/main/jni/freedcam -DANDROID -fopenmp -Wa,--noexecstack -Wformat -Werror=format-security -fopenmp -Wno-c++11-narrowing -std=c++11 -fpic -frtti -fexceptions     -isystem /usr/lib/android-ndk/platforms/android-14/arch-arm/usr/include -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/CustomMatrix.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/CustomMatrix.o 
```

I am using NDK r12b.

KillerInk, 2018-10-25:
oh sorry just use latest ndk 18.1.5063045. 
the readme is not up to date.
gdb got replaced by llvm and that makes ndk 12b no longer working.

KillerInk, 2018-10-25:
yes gnu_std support is also droppped from latest ndk

https://github.com/KillerInk/FreeDcam/blob/3de686b291c5ede065b1b75581a20fe00ac9143d/app/src/main/jni/Application.mk#L2

after that it should compile. its a bad moment to get it working^^

Rudloff, 2018-10-25:
I tried building 3de686b291c5ede065b1b75581a20fe00ac9143d with NDK r18b and I get this error again:
```
Execution failed for task ':app:generateJsonModelDebug'.
> Build command failed.
  Error while executing process /home/pierre/www/android-sdk/ndk-bundle/ndk-build with arguments {NDK_PROJECT_PATH=null APP_BUILD_SCRIPT=/home/pierre/www/FreeDcam/app/src/main/jni/Android.mk NDK_APPLICATION_MK=/home/pierre/www/FreeDcam/app/src/main/jni/Application.mk APP_ABI=armeabi-v7a NDK_ALL_ABIS=armeabi-v7a NDK_DEBUG=1 APP_PLATFORM=android-16 NDK_OUT=/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj NDK_LIBS_OUT=/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib -j4 APP_SHORT_COMMANDS=false LOCAL_SHORT_COMMANDS=false -B -n}
  Android NDK: WARNING:/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/Android.mk:raw: LOCAL_LDLIBS is always ignored for static libraries    
  Android NDK: WARNING:/home/pierre/www/FreeDcam/app/src/main/jni/tiff/Android.mk:tiff: LOCAL_LDLIBS is always ignored for static libraries    
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/*
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver
  rm -f /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  mkdir -p /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a
  echo [armeabi-v7a] "Gdbserver      ": "[arm-linux-androideabi] /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver"
  install -p /home/pierre/www/android-sdk/ndk-bundle/prebuilt/android-arm/gdbserver/gdbserver /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdbserver
  echo [armeabi-v7a] "Gdbsetup       ": "/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup"
  echo "set solib-search-path /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a" > /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  echo "directory /home/pierre/www/android-sdk/ndk-bundle/sysroot /home/pierre/www/android-sdk/ndk-bundle/sources/android/support /home/pierre/www/android-sdk/ndk-bundle/sources/android/support/include /home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++ /home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++abi/include null/tiff/libtiff/ /home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw /home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++/include /home/pierre/www/FreeDcam/app/src/main/jni/freedcam /home/pierre/www/FreeDcam/app/src/main/jni/tiff/libtiff" >> /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/lib/armeabi-v7a/gdb.setup
  mkdir -p /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam
  echo [armeabi-v7a] "Compile++ thumb": "freedcam <= RawToDng.cpp"
  /home/pierre/www/android-sdk/ndk-bundle/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/RawToDng.o.d -target armv7-none-linux-androideabi16 -fdata-sections -ffunction-sections -fstack-protector-strong -funwind-tables -no-canonical-prefixes  -g -Wno-invalid-command-line-argument -Wno-unused-command-line-argument  -fno-exceptions -fno-rtti -gcc-toolchain /home/pierre/www/android-sdk/ndk-bundle/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64 -fpic -march=armv7-a -mfloat-abi=softfp -mfpu=vfpv3-d16  -mthumb -O0 -UNDEBUG -fno-limit-debug-info  -Inull/tiff/libtiff/ -I/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw -I/home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++/include -I/home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++abi/include -I/home/pierre/www/android-sdk/ndk-bundle/sources/android/support/include -I/home/pierre/www/FreeDcam/app/src/main/jni/freedcam -std=c++11    -DANDROID -fopenmp -Wa,--noexecstack -Wformat -Werror=format-security -fexceptions -fopenmp -Wno-c++11-narrowing -frtti -std=c++11 -fpic -frtti -fexceptions  --sysroot /home/pierre/www/android-sdk/ndk-bundle/sysroot -isystem /home/pierre/www/android-sdk/ndk-bundle/sysroot/usr/include/arm-linux-androideabi -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/RawToDng.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/RawToDng.o
  echo [armeabi-v7a] "Compile++ thumb": "freedcam <= librawutils.cpp"
  /home/pierre/www/android-sdk/ndk-bundle/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ -MMD -MP -MF /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/librawutils.o.d -target armv7-none-linux-androideabi16 -fdata-sections -ffunction-sections -fstack-protector-strong -funwind-tables -no-canonical-prefixes  -g -Wno-invalid-command-line-argument -Wno-unused-command-line-argument  -fno-exceptions -fno-rtti -gcc-toolchain /home/pierre/www/android-sdk/ndk-bundle/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64 -fpic -march=armv7-a -mfloat-abi=softfp -mfpu=vfpv3-d16  -mthumb -O0 -UNDEBUG -fno-limit-debug-info  -Inull/tiff/libtiff/ -I/home/pierre/www/FreeDcam/app/src/main/jni/LibRaw/libraw -I/home/pierre/www/android-sdk/ndk-bundle/sources/cxx-stl/llvm-libc++/include -I/home/pierre/www/android-sdk/n [... cut ...] roid-sdk/ndk-bundle/sysroot/usr/include/arm-linux-androideabi -c  /home/pierre/www/FreeDcam/app/src/main/jni/freedcam/CustomMatrix.cpp -o /home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/CustomMatrix.o
  make: *** No rule to make target `/home/pierre/www/FreeDcam/app/src/main/jni/freedcam/Opcode.cpp', needed by `/home/pierre/www/FreeDcam/app/build/intermediates/ndkBuild/debug/obj/local/armeabi-v7a/objs-debug/freedcam/Opcode.o'.  Stop.
```

KillerInk, 2018-10-25:
for me it looks like that the Opcode.cpp is missing. im uploading my latest changes and add a new tag

Rudloff, 2018-10-25:
I think I found the problem. There is a typo in the `OpCode.cpp` filename here:
```diff
diff --git a/app/src/main/jni/freedcam/Android.mk b/app/src/main/jni/freedcam/Android.mk
index 876bdb811..e0a33310e 100644
--- a/app/src/main/jni/freedcam/Android.mk
+++ b/app/src/main/jni/freedcam/Android.mk
@@ -4,7 +4,7 @@ LOCAL_ARM_MODE := arm
 
 include $(CLEAR_VARS)
 LOCAL_MODULE    := libfreedcam
-LOCAL_SRC_FILES := RawToDng.cpp librawutils.cpp DngWriter.cpp DngStacker.cpp ExifInfo.cpp GpsInfo.cpp DngProfile.cpp CustomMatrix.cpp Opcode.cpp
+LOCAL_SRC_FILES := RawToDng.cpp librawutils.cpp DngWriter.cpp DngStacker.cpp ExifInfo.cpp GpsInfo.cpp DngProfile.cpp CustomMatrix.cpp OpCode.cpp
 LOCAL_CPPFLAGS := -fexceptions -fopenmp -Wno-c++11-narrowing -frtti -std=c++11 -fpic
 LOCAL_CFLAGS += -fopenmp
 LOCAL_LDFLAGS += -fopenmp
```

With this fix, it builds correctly :slightly_smiling_face:

KillerInk, 2018-10-25:
hm lol. i wonder why my ndk does not complain about. thx 😄

closed 2018-10-26 by KillerInk