KillerInk/FreeDcam #64: Could not find runtime.aar

Rudloff, 2018-10-24:
While doing test for https://github.com/KillerInk/FreeDcam/issues/63, I stumbled on this error:
```
Could not resolve all files for configuration ':app:debugCompileClasspath'.
> Could not find runtime.aar (android.arch.lifecycle:runtime:1.0.3).
  Searched in the following locations:
      https://jcenter.bintray.com/android/arch/lifecycle/runtime/1.0.3/runtime-1.0.3.aar
```

If I have `runtime-1.0.3.aar` from another Android project, it works correctly. But if I clean my Gradle cache, I always get this error when trying to build FreeDcam 4.1final.

For some reason, this seems to fix the issue:
```diff
diff --git a/build.gradle b/build.gradle
index 2b41db178..16e5041a7 100644
--- a/build.gradle
+++ b/build.gradle
@@ -15,8 +15,8 @@ buildscript {
 
 allprojects {
     repositories {
-        jcenter()
         google()
+        jcenter()
     }
 }
```

KillerInk, 2018-10-25:
i think it has something todo with latest gradle and the the build tools. 
with updating to as 3.1.2 and gradle 4.6 i was forced to update build tools to 28.0.2
i cant reproduce the problem, so i think the above is the solution.

my .idea/libraries/Gradle__android_arch_lifecycle_runtime_1_0_3.xml 

```
<component name="libraryTable">
  <library name="Gradle: android.arch.lifecycle:runtime-1.0.3">
    <CLASSES>
      <root url="file://$USER_HOME$/.gradle/caches/transforms-1/files-1.1/runtime-1.0.3.aar/5341938cd6b57d906bfa23a9917c5059/res" />
      <root url="jar://$USER_HOME$/.gradle/caches/transforms-1/files-1.1/runtime-1.0.3.aar/5341938cd6b57d906bfa23a9917c5059/jars/classes.jar!/" />
    </CLASSES>
    <JAVADOC />
    <SOURCES>
      <root url="jar://$USER_HOME$/.gradle/caches/modules-2/files-2.1/android.arch.lifecycle/runtime/1.0.3/4154878465bd9019f4c4988c668c7d69f2f687b3/runtime-1.0.3-sources.jar!/" />
    </SOURCES>
  </library>
</component>
```

Rudloff, 2018-10-25:
I tried building 3de686b291c5ede065b1b75581a20fe00ac9143d and I got a similar error:
```
> Could not resolve all artifacts for configuration ':classpath'.
   > Could not find aapt2-proto.jar (com.android.tools.build:aapt2-proto:0.3.1).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/build/aapt2-proto/0.3.1/aapt2-proto-0.3.1.jar
```

To fix it, I also had to change the order in `buildscript`:
```diff
diff --git a/build.gradle b/build.gradle
index 9b7402df8..ecd45f466 100644
--- a/build.gradle
+++ b/build.gradle
@@ -2,8 +2,8 @@
 
 buildscript {
     repositories {
-        jcenter()
         google()
+        jcenter()
     }
     dependencies {
         classpath 'com.android.tools.build:gradle:3.2.1'
@@ -15,8 +15,8 @@ buildscript {
 
 allprojects {
     repositories {
-        jcenter()
         google()
+        jcenter()
     }
 }
 ```

KillerInk, 2018-10-25:
normaly that stuff get downloaded on first project import. im a windows user, cant say much about osx or linux.
you can try to unload the project and reimport it. it fix most time such problems

Rudloff, 2018-10-25:
I don't use Android Studio, I run Gradle manually in a terminal. And that is also what F-Droid (https://github.com/KillerInk/FreeDcam/issues/8) does.

Would you accept a pull request with the small changes I mentionned above?
(This solution is also mentioned [here](https://github.com/invertase/react-native-firebase/issues/1189) and [here](https://stackoverflow.com/questions/50584437/android-studio-3-1-2-failed-to-resolve-runtime).)

KillerInk, 2018-10-25:
sure if it works for you. it help others too :D

closed 2018-10-25 by KillerInk