KillerInk/FreeDcam #67: Video file corrupts while recording on low free storage

aunqur, 2018-10-28:
Phone Model: Mi A1
---------------

FreeDcam Version: 4.1final
-----------------

Problem Description:
--------------------
If file save location of videos have low memory, It will be good to stop automatically while system have only 200-500 mb free space. Because the video recorded file will be corrupt and it still tries to write file when no storage space available.

Also with nearly zero memory the android internal os modules then fails to write settings on disk and automatically crashes ex:play services, system ui, etc.

Preserve some memory so system can work properly.

KillerInk, 2018-10-29:
thats a real problem. at current state freedcam ignores low storage space. 
dont know yet how to solve it.
maybe java have a callback wich notify about storage changes.
alternative a new thread must get started and check the every xx ms the storage size

aunqur, 2018-10-29:
There is a free storage detection that shows at right top along with current time and video profile. that detection can be used to trigger video stop at low storage.

closed 2018-10-30 by aunqur