KillerInk/FreeDcam #74: underexpose on tap to focus while recording video

aunqur, 2018-11-01:
Phone Model: MI A1
---------------

FreeDcam Version: 4.1final
-----------------

Problem Description:
--------------------
Start recording a video then if I tap to focus, it will down the exposure and scene gets darker. After that every time I focus it is back to the normal exposure mode. 

it seems after start the video then first tap to focus triggers something to underexpose the scene.

KillerInk, 2018-11-01:
im not sure, non of my devices have that behavior.
i could have something todo with the metering, and when the recording starts the mediarecorder takes over the control and may decide to use a different metering.

on most devices with deactivated zsl the preview ignores the shutter speed and iso with active ae.
the preview has there only the need to show a bright picture equal if its noise aslong its able to focus and the user see the scene he/she is capturing.
with starting a recording the media recorder takes control over the ae and set a fix framerate/shutterspeed.
if the preview used now and different speed then the 25fps it may take a few frames till the new iso value is correct measured

aunqur, 2018-11-02:
I've tried various settings zsl on/off rdi on/off nothing happens. it is more visible in low light situations but also it flicks quickly first time focus after start recording a video in properly lit environment. may be there is some video parameters set on first focus after start recording.

aunqur, 2018-11-03:
But this was ok on 4.1beta9. there might be something triggered on first focus while recording video. Here I've shared a clip of video recording on latest build: 

4.1final:
https://drive.google.com/file/d/1-5rkMnpF2itp6xVJ_hrBO8Zj8cKPMCal/view?usp=drivesdk

Also see the old app 4.1beta9:
https://drive.google.com/file/d/1-9GjVCTHvs9jX020IpBE-Yj6aSPSMOV8/view?usp=drivesdk

4.1beta9 screen recording:
https://drive.google.com/file/d/1-F2JpaYtzpBMxYr4BDax70u3HPo5eNe6/view?usp=drivesdk

You can see the difference while tap to focus. On focus event the camera will slightly defocus and then focus.

I am recording with 2 different build on same device and same settings.

aunqur, 2018-11-03:
I've investigated the Camera parameters before and after focus start. everything is same. only change is "touch-aec" on and "touch-index-af". I've tried without setting "touch-aec" but same result.

so i've deep down to video recording start call "recorder.start();" at freed/cam/apis/camera1/modules/AbstractVideoModule.java and checked the camera parameters. If I set the Camera parameters before the recorder.start() nothing happens and same result. but If i set video parameters immediately after recorder.start() it resolves the issue. I had to set "cameraUiWrapper.getParameterHandler().SetParameters();" two times as first call will under expose and 2nd call will make it normal.

Now if I tap to focus previous issue is gone and Exposure remains same on every tap of focus. It is really strange to me as I don't know the full project. 

May be some new parameters are causing this problem that is not available in 4.1beta9.

KillerInk, 2018-11-05:
you can dump parameters like this.

https://github.com/KillerInk/FreeDcam/blob/b193b29900e0d3b5c95b43399835645455f8ca3d/app/src/main/java/freed/cam/ui/themesample/settings/childs/SettingsChildMenuSaveCamParams.java#L131-L133

its not only strange for you, i dont see the problem. also no new parameters are added to camera1.

what happen when you touch to focus bevor you start recording?

KillerInk, 2018-11-05:
on left side in camera ui you see AE. what happen when you change it from` frame-avarage` to `center-weighted` or` spot-metering`?

aunqur, 2018-11-05:
Before recording start there is no effect on touch focus. Exposure remains same as it should be. 

Ive changed various AE mode it is working expected before recording. 

Only it happens after video recording start. Strange thing is if i lock the exposure before start recording it also change exposure after first focus. 

I'll cross check parameters with previous version. May be there is some parameters miss match.

KillerInk, 2018-11-05:
i dont think you will find a difference in the parameters between the versions.
as said, it works on p9, nexus5, lg g3, sammy s6/7 and good old optimus3d.

what you can try is to disable the qcom focus, maybe its something wrong with.

https://github.com/KillerInk/FreeDcam/blob/b193b29900e0d3b5c95b43399835645455f8ca3d/app/src/main/java/freed/cam/apis/camera1/parameters/ParametersHandler.java#L499-L501

aunqur, 2018-11-29:
I've checked with only setAndroidFocus(focusAreas); but the result is same. It happens on first SetParametersToCamera(cameraParameters); after start video record.

closed 2020-12-03 by KillerInk