KillerInk/FreeDcam #77: Samsung Galaxy S3 Duos (GT-I9300I) raw2dng problem

SPoket, 2018-11-15:
Phone Model:
Samsung Galaxy S3 Duos (GT-I9300I)

FreeDcam Version:
4.1final

Problem Description:

Hi! When I try to use raw2dng, i see this problem (and toast "Unknown raw"):

[image]

Here is my parameters:
[GT-I9300I_CameraParameters.txt](https://github.com/KillerInk/FreeDcam/files/2585613/GT-I9300I_CameraParameters.txt)

KillerInk, 2018-11-15:
When i remember correct s3 dont support raw. All trys ended in a jpeg file.
Unknow raw means that its not listed in the dngprofiles.xml. Its unknown to freedcam. Pls upload a bayer sample

SPoket, 2018-11-15:
Ok, bayer file:
[2018_11_15_18_55_06.zip](https://github.com/KillerInk/FreeDcam/files/2585808/2018_11_15_18_55_06.zip)
>When i remember correct s3 dont support raw. All trys ended in a jpeg file.
But .bayer generating works, & S3/S3 Duos different devices

KillerInk, 2018-11-15:
but its sammy. check on your own its a jpg. same problem as on the s3.