KillerInk/FreeDcam #86: Moto G 2014 XT1069 DNG 0 bytes/Crash

Fabio3rs, 2018-12-12:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt" - **not found**

Phone Model:
---------------
Moto G 2014 XT1069 16GB

FreeDcam Version:
-----------------
4.2

Problem Description:
--------------------
Crash or generate DNG with 0 bytes.
Bayer output is ok.

[Moto G 2014_CameraParameters.txt](https://gist.github.com/Fabio3rs/a1278235a5a1dde0d2272274648b88c9)
[Bayer file 2018_12_12_17_01_51.7z](https://drive.google.com/file/d/1bvRWeMIIez39rOyVRgpKJ2IFpr_Jm53S/view?usp=sharing)

KillerInk, 2018-12-13:
Hi, the attached bayer is in real a jpeg. 
```
picture-format-values=jpeg
picture-format=jpeg
```
without modded libs that device only support jpeg

Fabio3rs, 2018-12-13:
> Hi, the attached bayer is in real a jpeg.
> 
> ```
> picture-format-values=jpeg
> picture-format=jpeg
> ```
> without modded libs that device only support jpeg

Hi, thank you.

Which libraries exactly do I need to find modded or modify?

Fabio3rs, 2018-12-14:
I will look in XDA. Thank you.

closed 2020-12-03 by KillerInk