KillerInk/FreeDcam #87: Mi A2 lite not supported

letkan, 2018-12-22:
Phone Model:
---------------
Xiaomi Mi A2 lite
FreeDcam Version:
-----------------
4.2
Problem Description:
--------------------
This device doesn't seem to be supported ATM (no DNG and very few controls in Camera2) please tell me what info you need to get it going. I've attached camera parameters file.
[Mi_A2_lite_Phh-Treble vanilla_CameraParameters.txt.zip](https://github.com/KillerInk/FreeDcam/files/2704915/Mi_A2_lite_Phh-Treble.vanilla_CameraParameters.txt.zip)
Other then that the latest build is stable and works really good. Thanks.

KillerInk, 2018-12-22:
Hi xiaomi does not activate camera2. You have to root your device and enable it in the build.prop.without its only legacy device with a few options and you have more fun with camera1.

letkan, 2018-12-22:
I must correct myself, I reinstalled and I can get DNG with camera1, will let you know how it works with camera2 also. Sorry for the noise.

KillerInk, 2018-12-23:
no problem, 
after you applied 
`persist.camera.HAL3.enabled 1`
you have to rerun the feature detector from freedcam, else freedcam use the values for a legacy device.
all devices since L support camera2, but not with the full feature set.

https://source.android.com/devices/camera/versioning

closed 2020-12-03 by KillerInk