KillerInk/FreeDcam #89: Dng stack force closes

1400700226, 2018-12-27:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
Xiaomi mi max and mi8
FreeDcam Version:
-----------------
4.2
Problem Description:
--------------------
when I enter the dngstack screen, there is an RGB color "stack" picture, then I tap the picture, then freedcam force closes. Any help?

KillerInk, 2019-01-14:
hi
thats problem is caused by libtiff and the new ndk. its nolonger able to read the matrix stuff from the input dng. thats why its crashing. 
its also obsolete in meantime. in the dngstack-prebuilt brunch is a new approach for stacking dngs based on [this](https://github.com/KillerInk/hdr-plus). but its not yet at a stable state. whatever stable means in freedcam case^^

1400700226, 2019-01-15:
> hi
> thats problem is caused by libtiff and the new ndk. its nolonger able to read the matrix stuff from the input dng. thats why its crashing.
> its also obsolete in meantime. in the dngstack-prebuilt [...]

I know. Hoping the new version. Thanks

1400700226, 2019-01-15:
Oh... The close means close the issue.. Mistake sorry