KillerInk/FreeDcam #90: video upside down

letkan, 2019-01-01:
Phone Model:
Xiaomi MiA2 lite
FreeDcam Version:
4.2
Problem Description:
--------------------
While in picture mode I can turn the phone either right or left for landscape view, in video mode flipping it to the right takes the video upside down. Turning it to the left it shoots correctly. Is that a bug or what?

letkan, 2019-01-05:
> happy new year. happen that on camera1 or camera2?

camera1
Also, it doesn't want to do 120fps (slow motion), basically I just changed fps in 720p from 30 to 120 and turned off audio capture but the speed is the same.

KillerInk, 2019-01-05:
hm strange, must check it tomorrow, the problem with the orientation.

about highspeed make sure you set the record mode to highspeed. setting it to 120fps is not enough.
audio caputure should work with 120fps too

letkan, 2019-01-05:
Aha, thanks, will check it tomorrow.

closed 2020-12-03 by KillerInk