KillerInk/FreeDcam #92: [Bug] LG G6 ISO Manual controls broken

MegaArthur, 2019-02-09:
First of all, thank you for this amazing project, I see it as pretty much the best camera app, one that could even surpass GCam if HDR was implemented nicely.

The bug happens when I want to change the ISO manually on the LG G6 (AOSPExtended 5.8, android Oreo). It becomes extremely dark and doesn't really work.

KillerInk, 2019-02-10:
Hi myself only have a lg g3. G4 got burned. Cant say whats wrong for you.
On my nex5 is everything working. I tend to say its a lg specific problem.
Maybe you are so nice and tell me wich api you are using?
Greets

MegaArthur, 2019-02-10:
Camera2 API. It might be due to a custom rom

KillerInk, 2019-02-11:
yes could be a custom rom problem. but bevor you flash something else, you can try opencamera and see if its act the same way. if yes, its your rom. if no, its caused by freedcam.

MegaArthur, 2019-02-17:
OpenCamera's manual controls work fine

MegaArthur, 2019-06-27:
Hmm, it seems to _partially_ be a ROM issue. OpenCamera works, but the default camera has problems so it's probably not FreeDCam's fault

closed 2019-07-24 by MegaArthur