KillerInk/FreeDcam #95: Cropped liveview/viewfinder on Xiaomi Redmi Note 6Pro

viktorkrasovsky, 2019-05-07:
Phone Model:
Xiaomi Redmi Note 6 Pro

FreeDcam Version:
4.2

Problem Description:
The aspect ratio of the main camera is 4:3, but the liveview/viewfinder crops the picture to something wider like 3:2 or 16:9 (hard to tell). The worst thing that the liveview images is being cropped asymmetrically (it takes the upper part), so when I try to frame my subject it gets to the upper part of the actual picture and also gets distorted by lense's wide angle.

viktorkrasovsky, 2019-05-08:
I noticed when I switch to camera2 API then there's no crop, but I can't shoot RAW and do not have manual controls (probably because it works in legacy mode).

viktorkrasovsky, 2019-05-09:
See the screenshots. I notice that the cropping and aspect ratio is different between video and picture mode. The phone did not move while I was switching the modes. As you see in video mode the picture gets cropped correctly.
[image]
[image]

KillerInk, 2019-05-12:
looks for me as a different preview size get used.
for video the same size as the video have get used.
for picture its a bit different. there the size that match the picture aspect ratio get used <= 2560x1440.
on camera1 im not able to control if the preview gets cropped or a scaled down version from the full sensor is used. thats up to the oem.
thats different on camera2.

pls attach the camera parameters, open freedcam > settings > on left side scroll to bottom> click on save cameraparameters> you can find the file in internalSD/DCIM/FreeDcam/devicename_cameraparameters.txt

viktorkrasovsky, 2019-05-13:
Hi, here's the camera parameters file. I have also attached the the screenshot from Open Camera. It shows the correct aspect ratio of the preview (I guess it uses the camera1 API).
[Redmi Note 6 Pro_CameraParameters.txt](https://github.com/KillerInk/FreeDcam/files/3171790/Redmi.Note.6.Pro_CameraParameters.txt)
[image]

KillerInk, 2019-05-15:
from the paramters itself it looks good.
not sure whats going on

Bitte1Bit, 2020-04-16:
I've faced the same issue with Version 4.2 and try the older 4.1 because i naively expect, that with the bigger size of the apk-package (https://apkpure.com/freedcam/troop.com.freedcam) of that version, more functionality comes. And surprisesingly it works :-). May be the older version better fits to my old device, but thats also just a hypothesis.

MaferCJ, 2020-10-31:
Same bug on a Xiaomi Mi A2 using API1 (on API2 preview is ok, but API2 can only take long exposure photos up to 5s, while on API1 is up to 32s).

After digging more (and thanks to the hint of Bitte1Bit) I found the "culprit", commit [5b124ff](https://github.com/KillerInk/FreeDcam/commit/5b124ff0021667cacf2201961ed12fa60d3e2461).