KillerInk/FreeDcam #97: Interval mode with camera1 freezes app completely

Tooniis, 2019-08-07:
Phone Model:
Xiaomi Mi Note 2, MIUI Global 10.2.1.0(OADMIXM) Stable

FreeDcam Version:
4.2

Problem Description:
With `camera1` API, when shooting in interval mode, touching the shutter button takes 1 image then the application freezes completely until force stopped. It works with `camera2` API but it still causes a lot of stutter and it almost freezes yet it does not.

KillerInk, 2019-08-08:
yes there is something wrong

mvglasow, 2020-02-17:
Same here on a OnePlus One running LineageOS 15.1 (Android 8.1): with `camera1` (which the app selects by default) the app freezes in interval mode after taking one image. It does respond sporadically, though. Haven’t tried `camera2` yet.

On the other hand, the app stays responsive on my old Nexus S running KitKat (CyanogenMod 11), but Interval still takes only one picture.

mvglasow, 2020-02-17:
Tried `camera2` on the OnePlus One. Interval works but the app is not very responsive while image capturing is in progress. Looks like there is too much work being done on the UI thread which really should be offloaded to a background thread.

KillerInk, 2020-02-17:
its fixed in code already. but i dont find the commit.

mvglasow, 2020-02-17:
I’m on version 4.2_ (version code 142), as downloaded from F-Droid. Would the commit then be somewhere in https://github.com/KillerInk/FreeDcam/compare/4.2_...master ?

mvglasow, 2020-02-17:
Confirmed, d3a06ac on NexusS/CM11 works. Haven’t tried the OnePlus One yet.

closed 2020-12-03 by KillerInk