KillerInk/FreeDcam #99: Vertical line on raw images - LG G5

LukaHanzek, 2019-08-27:
Phone Model:
LG G5

FreeDcam Version:
4.2

Problem Description:
First of all, this is the best camera app I've seen but there is a major issue on LG G5 running Oreo (I don't know if the problem appears on other software versions). Raw images have some kind of line a little bit to the right of the center of the image. I have attached a low quality JPEG file where the issue can be seen clearly. JPEG images are fine and vertical line doesn't appear on them. The weird thing is there is no line on the images when viewing them through the FreeDCam Viewer which comes with the camera but as soon as I import the image into editing software or view it with default gallery app, the line appears. This app is the only option for shooting low light images (especially the 10bit RAW option) and I would be very thankful if this issue gets fixed.

[image]

KillerInk, 2019-08-28:
hi.
looks for me like a corrupt op code.
op code itself is a binary file wich contains mostly vignett correction and/or lens distortion correction.
That its not shown by the freedcamviewer is caused by libraw. it ignores opcode.
It should be the same for RawTherapee and DarkTable.
your captured dngs are not lost. you can remove the opcode with exiftools

When you are using camera2, you can activate the freedcam dng converter to workaround that problem.(Camera1 use is by default)
if the problem persists with freedcams dng converter, you can create your own opcode with shooting a whitewall and using it in lightroom with a plugin. 
In that case, the camera returns a bad gain map. its not freedcams fault^^
alternative you can use empty opcodes.
anyway both must placed in /FreeDcam/config/ 
depending on camera id you have to add the id infront. for back camera it would be 0opc3.bin, for front 1opc3.bin.

I attached the emtpy ones

[EmptyOPCodes.zip](https://github.com/KillerInk/FreeDcam/files/3550801/EmptyOPCodes.zip)

good luck

LukaHanzek, 2019-08-28:
Thank you very much for your help.
This was the problem indeed and putting empty op code files in config folder solved it. This camera app will be a fun way to learn some advanced things.

closed 2020-12-03 by KillerInk