KillerInk/FreeDcam #103: Remote control via USB (libgphoto2)

Stemby, 2020-01-03:
Hi,
I'd like to control the most common still camera in the world (an Android phone with integrated camera) with [Entangle](https://entangle-photo.org/).

Entangle uses libgphoto2:

http://gphoto.org/

Is it possible to expose FreeDcam features via USB in order to have remote control of an Android device from a GNU/Linux box (FOSS-only)?

Maybe by implementing [PTP](https://en.wikipedia.org/wiki/Picture_Transfer_Protocol#Device_control)?

Thank you!

Happy new year,
Carlo

KillerInk, 2020-01-04:
Hi,
Looks possbile todo, but i have not yet worked with USB.
Could get alot work work to handel all the commands.
Other problem is im not a linux user^^
But FreeDcam should be modular enough to handel it.
Streaming the preview could be tricky.

Happy new year too
greets

Stemby, 2020-01-05:
Hi @KillerInk,
thank you for your reply.

> Other problem is im not a linux user^^

This isn't a problem: PTP is a generic / neutral / cross platform protocol; gphoto2 should work automagically.

Best regards,
Carlo

nick87720z, 2020-02-24:
From own experience - I noticed, when android phone is connected in file view mode, both PC and phone software have access to files. E.g., when shooting some photos, they immediately appear in file manager at PC, if proper dir (DCIM/FreeDCam) is opened. This allowes 2 things:
- do long serious in multiple sub-serious, downloading each, when no enough free space at phone.
For example - you could do HDR serious, with e.g. 20 shots per each expo level (to do NR inside each sub-serie).
- side effect: preview files from PC without ever touching phone. This seems even more reliable than previewing in FreeDCam app. Because it seems, that when expo time is too big, no android photo app can display real result with such settings, because they never show live preview with such expo (or it would lower camera FPS to 1 and even below, what never happens).

Thus, the only remaining issue - adjusting controls remotely, to completely avoid touching smartphone during session.