KillerInk/FreeDcam #106: Redmi Note 8 Pro || force close

leon-nunes, 2020-02-25:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
No logs were found under this folder, hence I'm attaching Debug logs from ADB
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
Done
- Describe your Problem as good as possible.

Phone Model:
Redmi Note 8 Pro, Android 9, MIUI Global 11.0.4
---------------

FreeDcam Version:
4.2
-----------------

Problem Description:
As soon as the App is installed and you open it, the app FC's while running feature detector
[adb_debug_log.txt](https://github.com/KillerInk/FreeDcam/files/4250631/adb_debug_log.txt)

--------------------

KillerInk, 2020-02-26:
Is fixed with
https://github.com/KillerInk/FreeDcam/commit/543fceb09e70d631630271fdaafe86058a007b41

leon-nunes, 2020-02-26:
Hey,
Thanks for fixing this, I'm not that well versed with building Android apps or git cherry picking how would I download this build with a specific commit.
The release page is from months ago, should I build the app manually or is there another way thanks.

closed 2020-02-27 by leon-nunes

KillerInk, 2020-02-28:
when the saf is fully implemented and some other bugs are solved and not a new problem appears, a new release will roll out soon.

leon-nunes, 2020-02-28:
Hey,

I completely understand, I gave it a shot and built it.

Works like a charm thanks a lot for your help.