KillerInk/FreeDcam #108: Sporadically freezes in interval mode

mvglasow, 2020-03-01:
Today I did the endurance test for interval mode, with a 2s interval, shooting for 6 hours non-stop. At some point, however, FreeDcam simply froze and stopped responding.

### To reproduce:

1. Make sure you have ample free space on the device (for 5 MP at 95% quality, about 1 GByte/hr; it took me some 4.5h of shooting to run into the issue)
2. Select interval mode, 2s interval, repeat indefinitely.
3. Start taking pictures and wait.

### Expected result:

No fuss, one image every 2 (or so) seconds

### Actual result:

After about 4h 20min, FreeDcam stopped respondign altogether.

### Environment:

* FreeDCam@ dab91e8, built from source
* Nexus S running CyanogenMod 11 (Android 4.4.4)

### Further information:

Logcat here:
[alogcat.2020-03-01-16-45-27+0100.txt](https://github.com/KillerInk/FreeDcam/files/4272648/alogcat.2020-03-01-16-45-27%2B0100.txt)

The last image was taken at 16:26:22 The image looks OK; unfortunately the events had already rotated out of the log by the time I got around to saving it.

What seems to have happened: Apparently at some point the image did not get saved properly (hardware failure or just some kind of random hiccup in the OS—it happens with that kind of heavy usage). FreeDcam tries to read the picture it has just taken, which failed in this case. That sent FreeDcam into an endless loop of trying to re-read the picture. Having some kind of fail mechanism (e.g. skip the update this time, or show a blank image) would probably have solved this. In this case, we would lose one picture, but IMHO that is still better in a set-and-forget use case than the app stopping working altogether.

KillerInk, 2020-03-02:
sounds to me like the next race condition.
are the last stored images corrupt?
i think this could get fixed with reducing the pressure to the single core while a capture is in progress.
simplest solution is to keep the capture images in the module as long the interval runs, and on stop update the FileListController.

mvglasow, 2020-03-04:
The last image file does not appear to be corrupt. The image itself looks OK, and so does the EXIF information. I also did not notice any corruption in the preceding ones.

closed 2020-12-03 by KillerInk