KillerInk/FreeDcam #109: Improve interval reproducibility

mvglasow, 2020-03-01:
Experienced during the endurance test for interval mode, with a 2s interval, shooting for 6 hours non-stop:

### To reproduce:

1. Make sure you have ample free space on the device (for 5 MP at 95% quality, about 1 GByte/hr; it took me some 4.5h of shooting to run into the issue)
2. (Might no be needed, but was the case here) Run a processor-hungry background app on a weak device (in my case, a single-core CPU)
3. Select interval mode, 2s interval, repeat indefinitely.
4. Start taking pictures and wait.

### Expected result:

No fuss, one image every 2 seconds—unless the hardware can’t keep up, in which case I would expect the shortest interval the hardware can handle

### Actual result:

Minimum interval is around 3s, in my case it got as long as 6–7s after some 4 hours (though the delay may have been due to processor load caused by another app)

### Environment:

* FreeDCam@ dab91e8, built from source
* Nexus S running CyanogenMod 11 (Android 4.4.4)

### Further information:

I suspect FreeDcam does something like the following in interval mode:

1. Take a picture
2. Set timer to user interval
3. When timer fires, go back to 1.

This will always add the image taking/processing delay of the device to the actual interval. This could be improved as follows:

1. `time1 = timestamp`¹.
2. Take a picture.
3. `delay = timestamp - time1`
4. `if (delay <= interval) realInterval = interval - delay; else realInterval = 0;`
5. Set timer to `realInterval`
6. When timer fires, go back to 1.

¹) For `timestamp`, use a time value that is guaranteed to increase in a monotonous manner. Wall clock time will not work (as the clock might be changed by the user or via NTP), but system uptime should.

This should give intervals close to the user setting as long as the device can keep up. If the device cannot keep up, pictures are taken at the shortest interval the device can handle, without introducing any artificial delay. If this happens, using the timer with a zero (or near-zero) interval should ensure the app keeps responding to input.

mvglasow, 2020-03-14:
> sleep only 1000ms if sleepTimeBetweenCaptures > 0

Does that mean the app still sleeps for at least 1000 ms after taking a picture, even if taking the picture takes longer than the selected interval?

KillerInk, 2020-03-15:
no. that is added if you set the sleeptime to 0. bevor it sleept 1 sec also if it was set to 0
currently it works this way. 
on capture start it waits till its done. then the capture time gets substracted from the set sleeptime.
if the capture took longer then the sleeptime was set,next capture start immediately.
as sample you set exposuretime to 2 sec and interval sleep to 1 sec, its now continouse capturing.
lets use same expotime again but set sleeptime to 3 sec,its now sleeping 1sec to next capture

mvglasow, 2020-03-15:
Sounds good! I will report back after the next longer test drive, though I might not need interval mode for a while.

closed 2020-12-03 by KillerInk