KillerInk/FreeDcam #110: No location information in images

mvglasow, 2020-03-14:
### To reproduce:

1. Ensure '''Geo-Tag Location''' is on.
2. (Optional) Start an app which uses GPS in the background, to make sure we are constantly listening for location updates.
3. Go outside and start snapping pictures in interval mode.

### Expected result:

Geo-tagged pictures, at least when another GPS app has been active for some time

### Actual result:

No location information in pictures, even with another GPS app running in the background (and having a fix) for several hours.

### Environment:

* FreeDCam@ dab91e8, built from source
* Nexus S running CyanogenMod 11 (Android 4.4.4)

### Further information:

Luckily I had a GPS trace and was able to geotag the images afterwards. If you know it, there’s a workaround, but this behavior is annoying if you’re not expecting it.

KillerInk, 2020-03-15:
yes there is something wrong. also freedcam forget thats it was turned on. looks like it did not survive last rework.
but you can check if it have a lock. pull down the navbar and if you see the gps symbol it get applied to the img.

mvglasow, 2020-03-15:
In that case, I did have a fix, as another app on the device was recording a GPS trace. Also, the device was mounted on the dash of my car and shooting for several hours non-stop (and yes, GPS gets through my windshield).

Just ran a quick test with 70ffaf: FreeDCam registers for location updates when shooting in interval mode, it just appears that it is not processing the data it receives properly. (I didn’t wait to get a fix this time.)

KillerInk, 2020-03-17:
dont know works here as it should.
if it have a fix, getLastKnownLocation returns the locationdata.
after that the location gets updated every minute and if you moved 15meters.
[requestLocationUpdates](https://developer.android.com/reference/android/location/LocationManager#requestLocationUpdates(java.lang.String,%20long,%20float,%20android.location.LocationListener))

https://github.com/KillerInk/FreeDcam/blob/f19d59a515e17227b6bc5b51a687f6701b19e169/app/src/main/java/freed/utils/LocationManager.java#L73-L94

our use cases are different. i mainly use it to shot startrails or landscapes in a fix position.
never thought someone would use it that way and need realtime updates.

mvglasow, 2020-04-01:
No idea why it isn’t getting picked up, just some thoughts:

* I see no benefit in using `updateDistance` here: Presumably, in order to determine if we have moved beyond that distance, the device still needs to acquire a location, so the load on the system (including power consumption) is the same. The only difference is that certain location updates are not getting delivered to the app. That may be useful for an OpenStreetCam/Mapillary/etc. use case if you were to use the location update to trigger the camera, to avoid shooting more than one picture every 15 meters. Here, it may add up to 15 meters of error to the location.
* `getLastKnownLocation()` can be off by a large distance if the last location is old and/or you are in a moving vehicle. If the location supplies speed, a rough estimate of the error is `speed * age` (though, of course, acceleration, deceleration and turns would not be considered here). What is acceptable may depend heavily on the use case, but I would discard old locations.
* Also, for `getLastKnownLocation()`, preference is unconditionally given to GPS if both are available. However, a relatively fresh network location may be more accurate than a much older GPS location. Then again, network locations may be less accurate to begin with if the only reference is the nearest cell tower, though that should be reflected in the error. I’d choose between two locations based on a combination of age and error (or even average between the two).