KillerInk/FreeDcam #111: Crashes with java.lang.NoSuchMethodError: android.media.ExifInterface.<init>

mvglasow, 2020-03-16:
Occurs right after starting FreeDcam.

OS is Android 4.4.4, FreeDcam f19d59a. Seems to have been introduced by 7b2bbb4, as 70ffaf20 did not experience this issue yet.

Logcat: [alogcat.2020-03-16-22-49-34+0100.txt](https://github.com/KillerInk/FreeDcam/files/4340585/alogcat.2020-03-16-22-49-34%2B0100.txt)

KillerInk, 2020-03-17:
after checking the history it got added with
4a89dc528716deefb90d93c69ce70423527623f8

you removed the forced true from?
https://github.com/KillerInk/FreeDcam/blob/f19d59a515e17227b6bc5b51a687f6701b19e169/app/src/main/java/freed/file/FileListController.java#L62

on kitkat that uri stuff shouldnt get used

mvglasow, 2020-04-01:
ah… didn’t notice you’d disabled the version check again. I see 8de7a93 has it in place again; just tested that version and it seems to work again.

closed 2020-04-01 by mvglasow