KillerInk/FreeDcam #112: Vorbis is broken, Lossless Encoding Audio and custom profiles Please

Tall-Lobster, 2020-05-03:
Notes:  
------
- The Log is stored in: "InternalSD/DCIM/FreeDcam/log.txt"
- Give the Heading a usefull name like, "Huawei P9 does not record 4k"
- Describe your Problem as good as possible.

Phone Model:
---------------
LG G5

FreeDcam Version:
-----------------
Version 4.2 from F-Droid

Problem Description:
--------------------
Vorbis when i select this option for audio encoding I get the "prepare failed:prepare failed" message.

Also why is the frame rate for video capped at 30 fps when you go to settings? and you can't increase above 30 fps? I know my phone can handle 60 fps for 1080p quality but even though I change the recording to 60, it still says 30 fps in the settings. And if this is that the app is capped to 30 but it re-encodes the video in the desired fps, then please fix this.

Right now you can only encode audio in lossy formats, it would be great if you can encode audio in lossless formats such as FLAC.
Also I would like the ability to try out custom video + audio profiles

KillerInk, 2020-05-04:
Hi
some audio codecs works only with specific video containers. But its not granted to all codecs are implemented by the oem. and there is no way to check if its supported. you have to try and error.
Here is a [list](https://developer.android.com/guide/topics/media/media-formats)

The fps you see in settings is for timelapse recording.
Pls open the VideoProfileEditor and make there your fps changes.
Select the 1080 profile, rename it to 1080p60. change fps to 60 and save it.
Normaly then 60fps should work with your phone.

greets

Tall-Lobster, 2020-05-04:
Dear KillerInk:

I am so sorry I forgot to reply back right away to indicate that i got the 60fps and 120 fps working on my LG G5. Though there might be a bug for 720p60 since the app would crash but 120 fps was fine.
I got the 1080p60, 720p120, and 480p120 working.

And thanks to this app I finally have an open source alternative app that I can record in 4K and 1080p60fps.

Though it would be great if you can add Lossless Encoding such as FLAC which is available as mentioned here:
[image]

Wait now that I see that it is only supported for .flac, is it possible to program the app so that if they decide to record in flac that it creates a separate file with the flac track while the video would have a lossy compressed acc or any of the other functional lossy compressions?

Regards,

Tall Lobster

Tall-Lobster, 2020-05-04:
Dear KillerInk:

if you are dealing with another bug or something is getting in the way that would not allow you to implement the feature I suggested in the previous comment, could you then direct me to the location of the code so that I could contribute on my own time to add this feature?

Regards,

Tall Lobster

KillerInk, 2020-05-05:
about 720p60fps. could be that it dont support it. oems use predefined profiles wich limits resolutions to fps ranges.

in theory you can store all different formats int mkv. the question is, can the decoder read/play the file^^

freedcam use the mediarecorder. could be that you have to add you own mediacodec to merge the streams. i never tried it. 

there the audio codec get set:
https://github.com/KillerInk/FreeDcam/blob/c1bd877c62453ca8685d9e6c8d6ec62809818eac/app/src/main/java/freed/cam/apis/basecamera/record/VideoRecorder.java#L216

video codec:
https://github.com/KillerInk/FreeDcam/blob/c1bd877c62453ca8685d9e6c8d6ec62809818eac/app/src/main/java/freed/cam/apis/basecamera/record/VideoRecorder.java#L182

there the audio formats are hardcoded:
https://github.com/KillerInk/FreeDcam/blob/c1bd877c62453ca8685d9e6c8d6ec62809818eac/app/src/main/java/freed/cam/ui/videoprofileeditor/VideoProfileEditorFragment.java#L74-L80

start recording for cam2
https://github.com/KillerInk/FreeDcam/blob/c1bd877c62453ca8685d9e6c8d6ec62809818eac/app/src/main/java/freed/cam/apis/camera2/modules/VideoModuleApi2.java#L275

start recording cam1
https://github.com/KillerInk/FreeDcam/blob/c1bd877c62453ca8685d9e6c8d6ec62809818eac/app/src/main/java/freed/cam/apis/camera1/modules/AbstractVideoModule.java#L127

good luck

closed 2020-12-03 by KillerInk