KillerInk/FreeDcam #114: Pixel 2 XL Camera Crash

snajdovski, 2020-05-31:
Notes:  
------
```
2020-05-31 20:50:30.782 8826-8897/troop.com.freedcam E/AndroidRuntime: FATAL EXCEPTION: 020-05-31 20:50:30.769 8826-8897/troop.com.freedcam D/CameraHolderApi2: setOpModeForVideoStream method is not exist
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err: java.lang.NoSuchMethodException: android.hardware.camera2.CameraDevice.setVendorStreamConfigMode [int]
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at java.lang.Class.getMethod(Class.java:2072)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at java.lang.Class.getDeclaredMethod(Class.java:2050)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at freed.cam.apis.camera2.CameraHolderApi2.checkSetOpMode(CameraHolderApi2.java:86)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at freed.cam.apis.camera2.CameraHolderApi2.<init>(CameraHolderApi2.java:78)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at freed.cam.apis.camera2.Camera2Fragment.createCamera(Camera2Fragment.java:290)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at freed.cam.apis.basecamera.MainToCameraHandler.handleMessage(MainToCameraHandler.java:105)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:107)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at android.os.Looper.loop(Looper.java:214)
2020-05-31 20:50:30.769 8826-8897/troop.com.freedcam W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:67)
2020-05-31 20:50:30.771 8826-8897/troop.com.freedcam D/BackgroundHandlerThread: startBackgroundThreadCaptureSessionHandler
2020-05-31 20:50:30.778 8826-8826/troop.com.freedcam D/Camera2Fragment: SurfaceTextureAvailable
2020-05-31 20:50:30.778 8826-8897/troop.com.freedcam D/Camera2Fragment: Start Camera
    
    --------- beginning of crash
2020-05-31 20:50:30.782 8826-8897/troop.com.freedcam E/AndroidRuntime: FATAL EXCEPTION: CameraFragmentManager
    Process: troop.com.freedcam, PID: 8826
    java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
        at freed.cam.apis.camera2.Camera2Fragment.startCamera(Camera2Fragment.java:312)
        at freed.cam.apis.basecamera.MainToCameraHandler.handleMessage(MainToCameraHandler.java:86)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
2020-05-31 20:50:30.811 8826-8897/troop.com.freedcam I/Process: Sending signal. PID: 8826 SIG: 9

```
Phone Model:
---------------
Pixel 2 XL
FreeDcam Version:
-----------------
  versionName '4.2'
Problem Description:
--------------------
The App crashes instantly when compiled from source

KillerInk, 2020-06-01:
hi
i hope the comment fix your problem. but i dont think thats the real reason for the crash.
looks like the feature detector does not find all availible cams

snajdovski, 2020-06-01:
It did fix the issue!