KillerInk/FreeDcam #115: Add crash reporter

fabmazz, 2020-08-06:
First of all, thanks for the amazing work the developers have undertaken.

I wanted to try the app on a couple of samsung devices, but it just kept crashing without showing anything. Since it's not possible anymore to catch the stack trace within android (without root), and I'm not near any computer, I can't point to the error.

However, if a crash reporter library could be bundled with the app, that catches the crash and is able to send all the logs, I think it would be a huge improvement.
- Looking at other FOSS project, there are a couple of alternatives, for example ACRA (https://github.com/ACRA/acra).

KillerInk, 2020-08-06:
hi freedcam has a log/crash reporter.

create sdcard/DCIM/FreeDcam/log.txt
and freedcam starts dumping its log to that file.
make sure you removed freedcam from the open tasks. it check only once onCreate if the log.txt exists
when done with logging, dont forget to delete log.txt. else it grows continously^^

fabmazz, 2020-08-06:
Thanks for the reply!
It's good to know, however I still have to open the log, scan for the crash and send it. With a crash reporter this would be automatic. 
I understand that it requires some work, but users are generally lazy, if the app doesn't work, most people won't create the log file, and then comb through it and send the relevant part.