KillerInk/FreeDcam #118: Extra camera listed in Oneplus 6

AndySchroder, 2020-08-16:
Phone Model: Oneplus 6
---------------

FreeDcam Version: 4.3
-----------------

Problem Description:
--------------------

With version 4.3 I became available to select between multiple back cameras. However, I'm getting "Back 0", "Back 2", and "Back 3", but there are only 2 back cameras, not 3. It seems as though "Back 0" and "Back 3" are the same camera, but I can't seem to successfully take a picture with "Back 3", and "Back 3" seems to have a different shutter speed and parts of the image are blurry in the viewfinder unless you hold still for a while. Is "Back 3" some kind of experimental feature that is trying to merge information from "Back 0" and "Back 2"?

Note, front camera is named "Front 1".

KillerInk, 2020-08-16:
the new trend from oems is to hide cameras. freedcam must now check if a camera id returns parameters.
also there are now logical multi cameras. with that you can control hardware cameras that are linked to that logical camera.
in your case i expect that back3 can control back0 and back2.
normaly you should not see back3. freedcam dont support it.
looks like opo use there a hidden key to identfiy it wich is not included in the detection.

the front part is the camera facing. the number behind, the camera id used to open.

https://github.com/KillerInk/FreeDcam/blob/b841d4295f70c75967919f0301025831853beb7e/app/src/main/java/freed/cam/apis/featuredetector/Camera2FeatureDetectorTask.java#L515-L571

github-k8n, 2020-09-04:
@AndySchroder , Xiaomi Mi 8 here, i got also multiple "Back" Cameras despite only having 2 back cameras, but for example the "Back 4" seems to be the "Portrait" mode i remember from the "official" camera (long time ago before I installed lineageos) where it tries to make the background blurry like with a "real" camera :). Try to capture someting within about 1 meter and you'll see the surrounding part getting blurred (at least it does that on mine....

It would be nice if those camera selections could be enabled/disabled per camera (some might be not needed) and if you could **rename** them.. ("Front", "Back", "Portrait", "Zoom" etc).

(the "Front 5" is interesting, showing an IR picture of the front facing camera including IR illumination, not sure what that is normally being used, maybe face unlock in the dark?)

closed 2020-12-03 by KillerInk