KillerInk/FreeDcam #122: Clearer image in viewfinder than saved image.

AndySchroder, 2020-08-16:
Phone Model: Oneplus 6
---------------

FreeDcam Version: 4.2, 4.3
-----------------

Problem Description:
--------------------

Why does the viewfinder/preview give a clearer image in low light than the captured image? Doesn't matter what camera settings I use (DNG or JPG), I can get a better quality image by taking a screenshot of the viewfinder/preview. Seems like other camera programs do a similar thing too, but it doesn't make any sense.

AndySchroder, 2020-08-16:
Just tried this on a different phone model (a Motorola) and it doesn't seem like the issue is nearly as noticeable.

Not sure what causes this on the Oneplus 6, but it "seemed" like my old stand alone digital cameras had similar phenomenon, but never had the opportunity to do a screen shot on one of those to prove it to myself.

KillerInk, 2020-08-16:
not sure what it cause. could be different cases.

1. one possibility that it only looks like its clearer because the preview size is smaller then the captured jpeg.
2. other possibility is that the preview get a extra denosing stage. or by default use wavlet denoise for preview, also if its turned off.
3. somekind of pixelbinning is happening for preview.

i tend to say its the first case. my preview looks also cleaner, but i see noise inside wich is alot more noticable in the captured image.

AndySchroder, 2020-09-09:
1. Don't think so, because as I said, screen shot of the viewfinder is better than the image itself.

bsaunder2002, 2024-03-26:
I wonder if the act of pressing a button is causing the phone to move and the camera to blur.
I have a OnePlus 9 Pro and I get this a lot.  I press the button, the stupid shutter sound goes off, then after that, it records the photo.
(this is with the default cam ap)
I get tons of "I didnt take that blury mess photo of their knees" photos.

If I get a st/ir shutter release (I think I have an IR one for my dslr that may work with android), I'll setup a tripod and take some photos.  Then repeat holding it and pressing buttons.  Based on above, I bet I find that the camera in this phone does not show the last thing it recorded on the screen and write that to mem, instead waits for the button press then captures then writes.  Which may lead to the blur.